enjoy-digital/litex #1108: Generation of "mem_2.init" is non-deterministic

tcal-x, 2021-11-19:
Observed with the Arty A7-35T board.   It seems that `mem_2` is the storage for SoC CSRs.   

Each time I run `litex_boards/targets/digilent_arty.py`, a few bytes towards the end of `mem_2.init` change:
```
$ ./digilent_arty.py >& /dev/null
$ sum build/digilent_arty/gateware/mem_2.init
20800     1
$ ./digilent_arty.py >& /dev/null
$ sum build/digilent_arty/gateware/mem_2.init
05960     1
$ ./digilent_arty.py >& /dev/null
$ sum build/digilent_arty/gateware/mem_2.init
56127     1
```
The problem is that this small change in turn causes different behavior downstream in Yosys when using SymbiFlow.

enjoy-digital, 2021-11-19:
This is probably related to the identifier version (with the time of the build), you can disable it by setting `ident_version` to` False`: https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc_core.py#L102

tcal-x, 2021-11-19:
In this case, I needed to add the flag `--no-ident-version`:
```
$ ./digilent_arty.py --no-ident-version >& /dev/null
$ sum build/digilent_arty/gateware/mem_2.init 
54933     1
$ ./digilent_arty.py --no-ident-version >& /dev/null
$ sum build/digilent_arty/gateware/mem_2.init 
54933     1
```
Thanks!

closed 2021-11-19 by tcal-x