enjoy-digital/litex #1110: build/xilinx/symbiflow.py: Add a check that timing is met

tcal-x, 2021-11-23:
Currently the build template for using SymbiFlow does not check whether timing is met.   It should.

The `Makefile` that's generated sends the output from `symbiflow_route` (and each other step) to `/dev/null`.   If I generate the build directory from `./digilent_arty.py --toolchain=symbiflow --build`, and then capture the output from `symbiflow_route`, I see
```
sys_clk to sys_clk CPD: 15.2655 ns (65.5072 MHz)
```
which does not meet the 100MHz target.   It does happen to run on my particular board, but it's not guaranteed to run on all boards.

enjoy-digital, 2021-11-23:
Hi @tcal-x,

as a first step we should indeed at least redirect the output to log files (as it is currently done on others build backends). We could then provide an option to detect timings issues during the build and select if we want to generate a warning or an error (in some cases when doing development, you still want to be able to build the design that still have tiny timing violations).

mithro, 2021-11-23:
If timing is not met, the SymbiFlow tools should be failing with a non-zero exit code. Is that happening?

tcal-x, 2021-11-24:
It seems that `symbiflow_route` thinks that everything is fine even when it doesn't meet timing.  Here are some lines from the log (period is 10ns):

Note negative slack:
```
Final critical path delay (least slack): 20.7591 ns
Final setup Worst Negative Slack (sWNS): -10.7591 ns
Final setup Total Negative Slack (sTNS): -11485 ns

Final setup slack histogram:
[ -1.1e-08: -8.8e-09)   57 (  0.4%) |*
[ -8.8e-09: -6.7e-09)   19 (  0.1%) |
[ -6.7e-09: -4.7e-09)  662 (  4.5%) |**********
[ -4.7e-09: -2.7e-09) 1173 (  8.0%) |*****************
[ -2.7e-09: -7.3e-10) 1427 (  9.7%) |********************
[ -7.3e-10:  1.3e-09) 1026 (  7.0%) |***************
[  1.3e-09:  3.3e-09) 2124 ( 14.4%) |******************************
[  3.3e-09:  5.3e-09) 2300 ( 15.6%) |*********************************
[  5.3e-09:  7.3e-09) 2719 ( 18.5%) |***************************************
[  7.3e-09:  9.3e-09) 3204 ( 21.8%) |**********************************************
```

Only guaranteed to run at 48MHz, requested 100MHz:
```
  sys_clk to sys_clk CPD: 20.7591 ns (48.1716 MHz)
```

Final lines:
```
VPR succeeded
The entire flow of VPR took 278.13 seconds (max_rss 3562.2 MiB)
```

Status:
```
% echo $?
0
```

mithro, 2021-11-24:
Please log a bug against symbiflow about the above.