enjoy-digital/litex #1122: `litex_setup.py --install --user` fails when trying to install `migen`

gkelly, 2021-12-05:
While following `README.md` (https://github.com/enjoy-digital/litex/blob/master/README.md?plain=1#L145) the `litex_setup.py --init --install --user` invocation fails with:

```
[Installing migen]...
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --user not recognized
Traceback (most recent call last):
  File "/home/gkelly/p/butter/./litex_setup.py", line 275, in <module>
    main()
  File "/home/gkelly/p/butter/./litex_setup.py", line 261, in main
    litex_setup_install_repos(user_mode=args.user)
  File "/home/gkelly/p/butter/./litex_setup.py", line 141, in litex_setup_install_repos
    subprocess.check_call("python3 setup.py develop {options}".format(
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'python3 setup.py develop --user' returned non-zero exit status 1.
```

Without `--user` it seems to have completed installing into the active venv without much hassle.

nickoe, 2021-12-28:
It worked for me the other day, but I think the issue there being is something in your environment. Maybe run the setup.py command manually to check if there is more info?

dnltz, 2022-01-03:
Seems like Migen does not support Python 3.8+ right now.

```
$ python3.8 setup.py develop --user
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --user not recognized
$ python3.7 setup.py develop --user
running develop
running egg_info
writing migen.egg-info/PKG-INFO
writing dependency_links to migen.egg-info/dependency_links.txt
writing requirements to migen.egg-info/requires.txt
writing top-level names to migen.egg-info/top_level.txt
reading manifest file 'migen.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
...
```

My workaround was to modify the subprocess call (line 142) in litex_setup from `python3 setup.py ...` to `python3.7 setup.py ...`.

nickoe, 2022-01-29:
@dnltz For the record I am curremtly using python 3.10 on Arch.

enjoy-digital, 2022-04-22:
This is probably fixed with https://github.com/enjoy-digital/litex/pull/1248. (At least installation works with it on Ubuntu 20.04 with Python 3.8+).

closed 2022-04-22 by enjoy-digital