enjoy-digital/litex #1142: litex_sim: Using litex_term in bridge mode to flash boot.json fails with recv timeout

nickoe, 2021-12-24:
## Background 
I am trying to simulate my own SoC design based on the default litex_sim, aka  I only added a couple of custom modules, everything else should be the same, so I can also test this issue with normal litex_sim.

My goal being is to run with a bare bones application on the CPU and initialize some memory in sdram. On the target hardware this appears to work find with litex_term via serial, but in simulation it does not when I do it in this way:

## Steps to reproduce
```
litex_sim --with-etherbone --uart-name crossover --with-sdram
litex_server --udp --udp-ip=192.168.1.51
litex_term --images boot.json bridge
``` 

Perform above three commands in three terminal sessions.

## Observation

What I see it that litex_term appears to transfer my other test data ok, but when transferring the test demo.bin it fails as follows:
 ```
litex_term --images boot.json bridge 

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec 24 2021 14:19:51
 BIOS CRC passed (d3587dd9)

 Migen git sha1: ac70301
 LiteX git sha1: b4bb7aa7

--=============== SoC ==================--
CPU:		VexRiscv @ 1MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		65536KiB 32-bit @ 1MT/s (CL-2 CWL-2)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (8.0KiB)...
  Write: 0x40000000-0x40002000 8.0KiB   
   Read: 0x40000000-0x40002000 8.0KiB   
Memtest OK
Memspeed at 0x40000000 (Sequential, 8.0KiB)...
  Write speed: 291.7KiB/s
   Read speed: 276.4KiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading demo.bin to 0x40000000 (6868 bytes)...
[LXTERM] Upload calibration... (inter-frame: 10.00us, length: 64)
Process Process-2:==> | 99%
Traceback (most recent call last):
  File "/usr/lib/python3.10/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/nickoe/litex/litex/litex/tools/litex_term.py", line 126, in crossover2pty
    elif not self.rxempty.read():
  File "/home/nickoe/litex/litex/litex/tools/remote/csr_builder.py", line 40, in read
    datas = self.readfn(self.addr, length=self.length)
  File "/home/nickoe/litex/litex/litex/tools/litex_client.py", line 64, in read
    packet = EtherbonePacket(self.receive_packet(self.socket))
  File "/home/nickoe/litex/litex/litex/tools/remote/etherbone.py", line 369, in receive_packet
    chunk = socket.recv(header_length - len(packet))
TimeoutError: timed out
```

The `boot.json` looks like this.
```
$ cat boot.json 
{
	"test_data_signed.cs16":       "0x41000000",
	"demo.bin":       "0x40000000"
}
```

I tried to increase the timeout on the socket just before the `recvfrom` in `def read(...)` in `comm_udp.py`, but it still fails.
``` comm_udp.py
        self.socket.settimeout(5)
        datas, dummy = self.socket.recvfrom(8192)
```
## Additionally
Originally I would have assumed that I could just do:
```
litex_sim --with-sdram --sdram-init boot.json
``` 
But that just stalls after the liftoff message, while
```
litex_sim --with-sdram --sdram-init demo.bin
``` 
works OK, but it does not initialize the other memory region as _I wanted_. I am not sure if both symptoms are related or not.

enjoy-digital, 2021-12-24:
Can you try with `litex_term` in `--safe `mode?

nickoe, 2021-12-24:
I am not at my workstation now, but I also think I tried that in some combination, but I will certainly dounle check later.

nickoe, 2021-12-24:
@enjoy-digital Ok, so it do indeed appear to work with the `--safe` option to `litex_term`, but it is also way slower at doing the "uploading" stage.

nickoe, 2021-12-25:
Should I make another issue for part 2 of by issue?

enjoy-digital, 2022-03-29:
@nickoe: Initializing memory from `.json` files is now supported by `litex_sim`.

closed 2022-03-29 by enjoy-digital

nickoe, 2022-03-29:
@enjoy-digital Ohh, cool, I will try it out soon. It will be quite helpful :)