enjoy-digital/litex #1146: Running the example lab004 in arty board.

controlpaths, 2021-12-27:
hello, I am trying to run the example in the lab004 in the Arty board but when I execute the make file there are some erros. First of all, I have copied the firmware folder to litex-boards/litex_boards/targets, then I have modify the make file BUILD_DIR to
`BUILD_DIR=../build/digilent_arty`
if I execute the make file, I have some warning of implicit declaration of some fuctions and unrecognised emulation mode.

~/litex/litex-boards/litex_boards/targets/firmware$ make all
 CC       main.o
main.c: In function 'display_test':
main.c:88:3: warning: implicit declaration of function 'display_sel_write'; did you mean 'display_test'? [-Wimplicit-function-declaration]
   display_sel_write(i);
   ^~~~~~~~~~~~~~~~~
   display_test
main.c:89:3: warning: implicit declaration of function 'display_value_write'; did you mean 'display_test'? [-Wimplicit-function-declaration]
   display_value_write(i);
   ^~~~~~~~~~~~~~~~~~~
   display_test
main.c:90:3: warning: implicit declaration of function 'display_write_write'; did you mean 'ddrphy_rst_write'? [-Wimplicit-function-declaration]
   display_write_write(1);
   ^~~~~~~~~~~~~~~~~~~
   ddrphy_rst_write
 LD       firmware.elf
riscv64-unknown-elf-ld: unrecognised emulation mode: abi=ilp32
Supported emulations: elf64lriscv elf32lriscv
make: *** [Makefile:18: firmware.elf] Error 1

regarding the fail of the emulation mode, I notice that the make file calls the -ld linker. I can fix this issue by calling the -gcc compiler by changing this line
`	$(LD) $(LDFLAGS) \`
by this
`	$(CC) $(LDFLAGS) \`

Executing again the make file, there are a missing file named crt0.o

riscv64-unknown-elf-gcc: error: ../build/digilent_arty/software/libbase/crt0.o: No such file or directory

I check this file in other folder and it is the definition of the _start vector. I can point to the this file in other folder, but then other errors appears.

Is there anything that I am doing wrong?  The compiler I am using is the download with the litex function.

Thanks!

enjoy-digital, 2021-12-27:
Hi @controlpaths,

`lab004 `was done for the Nexys DDR4 target present in `fpga_101` repo, so would probably need to be adapted to run on Arty.

You should probably try the LiteX bare metal demo instead, available here: https://github.com/enjoy-digital/litex/tree/master/litex/soc/software/demo

controlpaths, 2021-12-27:
Many thanks! it works at the first try. With this example I have a base.

Greetings.

closed 2021-12-27 by controlpaths