enjoy-digital/litex #1147: SDCard cannot be accessed from ECP5 and ESP32 on ULX3S v. 3.x (fix included)

BrunoLevy, 2021-12-27:
The SDCard is wired on both the ECP5 and the ESP32 on the ULX3S. 
- To access it from the ECP5, the ESP32 needs to be switched off (pin "F1" ("wifi_en") needs to go low)
- To access it from the ESP32,  clk, cs_n and mosi need to go high-impedence

Here is a way of connecting it either to the ECP5 (default) or to the ESP32 based on a CSR:
_(disclaimer: I'm no LiteX/nMigen/Python expert ! There may be cleaner ways of doing that)_

1) `litex-boards/litex_boards/platforms/radiona_ulx3s.py`, add pin definition
```
    ("wifi_en", 0, Pins("F1"), IOStandard("LVCMOS33"), Misc("PULLMODE=UP"), Misc("DRIVE=4")), 
```

2) `litex/litex/soc/integration/soc.py`, add three-state mode for `clk`, `cs_n` and `mosi`:
``` 
    def add_spi_sdcard(self, name="spisdcard", spi_clk_freq=400e3, software_debug=False):
        # Imports.
        from litex.soc.cores.spi import SPIMaster

#        pads = self.platform.request(name)
#        if hasattr(pads, "rst"):
#            self.comb += pads.rst.eq(0)

        # Pads.
        pads = Record([('clk', 1), ('cs_n', 1), ('mosi', 1), ('miso', 1)])

        # Core.
        self.check_if_exists(name)
        spisdcard = SPIMaster(pads, 8, self.sys_clk_freq, spi_clk_freq)
        spisdcard.add_clk_divider()
        setattr(self.submodules, name, spisdcard)

        # Tristate
        from migen.fhdl.specials import Tristate                
        self.spi_sdcard_oe = Signal()
        spi_pads = self.platform.request(name)
        self.specials += Tristate(spi_pads.clk,  pads.clk,  self.spi_sdcard_oe)
        self.specials += Tristate(spi_pads.cs_n, pads.cs_n, self.spi_sdcard_oe)
        self.specials += Tristate(spi_pads.mosi, pads.mosi, self.spi_sdcard_oe)
        self.comb += pads.miso.eq(spi_pads.miso)
        
        # Debug.
        if software_debug:
            self.add_constant("SPISDCARD_DEBUG")
```
3) `litex-boards/litex_boards/targets/radiona_ulx3s.py`: add CSR and wire it to `wifi_en` and `spi_sdcard_oe`:
```
class ESP32(Module, AutoCSR):
    def __init__(self, platform):
       self._enable = CSRStorage()
       self.comb += platform.request("wifi_en").eq(self._enable.storage)
```
and in BaseSOC constructor:
```
        self.esp32 = ESP32(platform)        
        self.submodules.esp32 = self.esp32
```
in `main()`
```
    if args.with_spi_sdcard:
        soc.add_spi_sdcard()
        soc.comb += soc.spi_sdcard_oe.eq(~soc.esp32._enable.storage)
```

enjoy-digital, 2022-01-14:
Thanks @BrunoLevy, with https://github.com/enjoy-digital/litex/commit/96a9495d6a499da50470da777036a7f6feef7301 you can now set `with_tristate=True` and use the `spisdcard_tristate` signal to control tristate behaviour.

enjoy-digital, 2022-03-29:
I think we can now now close this. SDCard has also been tested recently in SD-Mode on the ULX3S.

closed 2022-03-29 by enjoy-digital