enjoy-digital/litex #1149: Zynq7000: #1141 follow up improvements.

enjoy-digital, 2022-01-03:
- Move `linker-zynq.ld` to cores/cpu/zynq7000 and rename to `linker.ld`.
- Add linker attribute to `CPU` that defaults to `linker.ld`
- Make Builder/Makefile uses the linker defined by the attribute and first search it in the CPU directory, if not present, search it in the BIOS directory (will be the default).
- Move` LSCRIPT` to `soc/integration/export.py` instead of the #ifdef in the Makefile.

sergachev, 2022-01-03:
Besides the linker script - I'm thinking of ways to include manufacturer HAL/BSP C libraries to enable software compilation for hard CPU cores. So far I've seen 3 levels of complexity:
 - Xilinx Zynq: pretty easy - one can clone https://github.com/Xilinx/embeddedsw and compile the necessary files using own scripts, like https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/digilent_zedboard.py#L108 - but probably this recipe should be moved from boards repo into main litex
 - Quicklogic EOS S3: sources can be gathered in https://github.com/QuickLogic-Corp/qorc-sdk/ but probably have to be patched before compilation because for instance unnecessarily include FreeRTOS somewhere. I'm working on a demo project.
 - Gowin EMCU: sources are in zip files on manufacturer's site behind login :neutral_face: . Looks annoying, sure one can script archive verification and unpacking, but ...
 
Rewriting them from scratch is doable too, but will take some time.

sergachev, 2022-01-03:
Also very likely
 - same linker script can be used for most ARM cores
 - using includes large part of these can be made common for all cores