enjoy-digital/litex #1150: Documentation flag "--doc" not working

zeeshandildar-rs, 2022-01-04:
I was trying to generate document along with the soc but unfortunately "--doc" is not working at the recent commit " f04a3dcc50aab46a86067b04bf2eb9284cd3a89d". Kindly resolve this issue as soon as possible.

enjoy-digital, 2022-01-05:
This is still working correctly here. Issue with Python 3.10 has been reported on IRC, which version of Python are you using? (LiteX hasn't  been tested with 3.10 yet).

zeeshandildar-rs, 2022-01-05:
I am using python version 3.6.8

enjoy-digital, 2022-01-05:
OK, sorry I'm not able to help with the provided information. Please have a look at https://github.com/enjoy-digital/litex/wiki/Feedback-Contribution-Support#how-to-create-an-issue-when-it-does-not-work and provide the minimal information required to be able to help/reproduce.

zeeshandildar-rs, 2022-01-05:
I have cloned the litex github repository and build the soc using following command:
`lxsim --cpu-type=serv --doc`
It successfully built the soc but there is not "doc" directory in "/build/sim/" so I can't see the documentation generated along with this soc.

enjoy-digital, 2022-01-05:
`lxsim --cpu-type=serv --no-compile-gateware --doc` correctly generates the doc on my machine, can you test this? If not I'll need to build logs.

zeeshandildar-rs, 2022-01-06:
Yes it is working now. So, it is compulsory to add --no-compile-gateware along --doc. right?

enjoy-digital, 2022-01-07:
The documentation can only be created once the SoC has been build. So in simulation we either have to avoid running the simulation (`--no-compile-gateware`) or run it but do a CTRL+C to interrupt it to generate the doc.

closed 2022-01-07 by enjoy-digital