enjoy-digital/litex #1152: led blink example

suarezvictor, 2022-01-06:
hi! I friend of mine challenged me saying that if LiteX/nMigen makes things so easy, I should be show him an example of a led blink project in less than 10 lines. I did that, as follows:

`from migen import Module, Signal
import litex_boards.platforms.de0nano as board # or "arty" for Xilinx, etc.

platform = board.Platform()
led = platform.request("user_led", 0)   #board pin to use

module = Module()
counter = Signal(26)                    #declare 26-bit register
module.comb += led.eq(counter[25])      #led = highest bit
module.sync += counter.eq(counter + 1)  #increment counter

platform.build(module)                  # Build bitstream
`

The thing is that I expected an example like that in the wiki or somewhere in the docs, but couldn't find it. Is it there? If not this one can be used.

By the way when you enter to the wiki page https://github.com/enjoy-digital/litex it circularly references to itself when you actually expect mode details...

suarezvictor, 2022-01-07:
Here an example using custom verilog:
```
verilog = """
module blink(output led, input clk);

reg [26:0] counter;
always @(posedge clk)
  counter <= counter+1;
assign led = counter[26] & counter[24];

endmodule
"""

import litex_boards.platforms.de0nano as board # or "arty" for Xilinx, etc.
platform = board.Platform()
pin = platform.request("user_led", 0)   #board pin to use

from migen import *
with open("module.v", "w") as f:
    f.write(verilog)
platform.add_source("module.v")
top = Module()
top.specials += Instance("blink", o_led=pin, i_clk = ClockSignal("sys")) #i_rst = ResetSignal("sys")
platform.build(top)

```

enjoy-digital, 2022-01-07:
Thanks @suarezvictor, it would indeed be nice to have a such example to the repo. In fact, this is similar to the Quick Intro from Migen: https://github.com/m-labs/migen#quick-intro (Note that LiteX use Migen, not nMigen and that nMigen used by the community is now named Amaranth). Some work is planned to improve the wiki this year.

suarezvictor, 2022-01-07:
I've seen the nice tutorials you referenced but theyw were hard to reach for me. Hopefully a "hello world" example like this is easier to find in LiteX docs ;)
I didn't know Migen and nMigen were different tools...

enjoy-digital, 2022-01-07:
Sure yes, that's a good idea and interesting feedback. (When you are the developer of a project and very familiar with it more difficult to see this :)).

developandplay, 2022-02-05:
@suarezvictor I found walking through the litex fpga-101 the best resource when initially getting started: https://github.com/litex-hub/fpga_101

Also +1 for the Migen/nMigen confusion. Guess I never used nMigen then...