enjoy-digital/litex #1163: Using LiteXTerm from Python rather than using the lxterm script

noorassan, 2022-01-15:
Hi! I was wondering if somebody could point me in the right direction as to how I can use LiteXTerm (from litex_term.py) directly from Python rather than through invoking the `lxterm` bash script. My goal is to upload a RISCV binary to my board (Arty A7), which I've already programmed with a bitstream, and boot the SoC. It seems like it should be a fairly straightforward process, but none of my attempts so far have given me entirely satisfactory results. Note that my goal is to keep the RISCV program running – my understanding is that the `lxterm` bash script stops the program once it is exited.

I've tried the following:

       def __run_riscv_program(self):

            self.lxterm = LiteXTerm(True, self.config["RiscVBinary"], self.config["KernelAddr"], None, True)

            self.lxterm.open(self.config["DeviceFile"], self.config["UartSpeed"])

            self.lxterm.console.configure()

            self.lxterm.start()

            self.lxterm.console.unconfigure()

This gets the job done. However, it severely messes with stdin. When I invoke this method from the Python console, I'm basically unable to type in any further commands.

I'm guessing that I'm missing a fairly basic error here or perhaps entirely misusing LiteXTerm altogether, but if somebody could point me in the right direction, it would be much appreciated.

noorassan, 2022-01-15:
Forgot to add – I've also the following:

        def __run_riscv_program(self):
            self.lxterm = LiteXTerm(True, self.config["RiscVBinary"], self.config["KernelAddr"], None, True)
            self.lxterm.open(self.config["DeviceFile"], self.config["UartSpeed"])
            self.lxterm.upload(self.config["RiscVBinary"], int(self.config["KernelAddr"], base=16))
            self.lxterm.boot()

Which results in output like the following:

        [LXTERM] Uploading ../../build/software.bin to 0x40000000 (4156 bytes)...
        [LXTERM] Got unexpected response from device 'b'\xa2''

enjoy-digital, 2022-01-16:
Hi @noorassan,

If you want to re-configure a memory of the SoC directly from Python, you could also use one of the [bridges](https://github.com/enjoy-digital/litex/wiki/Use-Host-Bridge-to-control-debug-a-SoC), open litex_server and a script similar  to [this](https://github.com/enjoy-digital/litedram/blob/master/bench/common.py#L95-L109) .

On Arty, you could operate the bridge over Etherbone (`--with-etherbone`) or JTAG (`--jtagbone`) and keep the regular UART. 
The UART has a specific behaviour on Arty and is reseted when connection with Host is opened/closed, to avoid this you can set `with_rst=False `[here](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/digilent_arty.py#L33).

noorassan, 2022-01-17:
Hi, @enjoy-digital,

Thanks for the guidance – this seems like it'll be a much better fit for me than going through litex_term. 

As you suggested, I'm using litex server over JTAG along with the script for writing the binary to the SoC. I also set with_rst=False.  It works fine up until I write to the ctrl_reset register, but after making that write, other reads/writes to the SoC all hang and then fail with a socket timeout. Do you have any idea what might be going wrong? Thanks!

enjoy-digital, 2022-01-17:
Hi @noorassan,

the JTAG core is also probably reseted by the write to ctrl_reset_register.  I would recommend commenting [this line](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/digilent_arty.py#L47) which will disable the reset of the peripherals and only keep the reset of the CPU.

noorassan, 2022-01-19:
@enjoy-digital 

Thanks! That did fix the issue I described -- reading/writing is successful after the write to ctrl_reset. I just want to confirm that the expected behavior of writing 1 to ctrl_reset is for the SoC to begin running from ROM -- is that correct? In my case, it doesn't seem that anything happens after I write to ctrl_reset (I don't see any change in the CSRs that my code writes to). Is there any documentation on how ctrl_reset is supposed to work?

Thanks so much for your help!

noorassan, 2022-01-23:
Ah! It turns out that my ROM was defined as read-only, so none of the writes I was making to program the SoC with the binary were succeeding. Changing my ROM to be writable resolved the issue (though it does seem a little wrong for ROM to be writable to, so I wonder if there's a better solution here). Nonetheless, I'll mark this as resolved. Thanks for the help!

closed 2022-01-23 by noorassan