enjoy-digital/litex #1164: Ethernet on Colorlight 5a-75b V8.0

faeboli, 2022-01-16:
Hello,
I'm starting to explore litex on ECP5 using a colorlight 5a-75b V8.0 board.
I suppose I'm having a problem with ethernet:
litex 2021.12
Step1:
if I build base target with options: ./colorlight_5a_75x.py --board=5a75b --revision=8.0 --build
then flash the .bit file generated under gateware folder using openFPGAloader trough a FT232h dongle,
I get correctly the bios output on uart.
Step2:
If I perform the same steps adding ethernet: ./colorlight_5a_75x.py --board=5a75b --revision=8.0 --build --with-ethernet
the build completes successfully, but I'll not get any text on uart. Also ethernet does not show any activity.

Is this normal behavior?
What else can I do next to understand what is going on?
Thanks

enjoy-digital, 2022-01-16:
Hi,

sorry I don't have the hardware, but this could be related to the Ethernet reset. Can you comment the `rst_n` pin on Ethernet: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/colorlight_5a_75b.py#L244
And see if you now get the UART?

faeboli, 2022-01-16:
Thank you very much!
Commented the line and now uart is showing BIOS messages, ethernet is alive also.

enjoy-digital, 2022-01-17:
@faeboli: Good, thanks for the test and feedback. This is fixed with https://github.com/litex-hub/litex-boards/commit/b240c381a7b60f31b74a784c3574f7f1c12b5022.

closed 2022-01-17 by enjoy-digital