enjoy-digital/litex #1168: No output while running demo.bin on Litex SoC with Rocket CPU hardware(Xilinx AC701 board) after Liftoff!

Aravindh-Swaminathan, 2022-01-18:
Generation of bit stream.
` litex-boards/litex_boards/targets/xilinx_ac701.py --build --cpu-type rocket --cpu-variant linux4 --sys-clk-freq 50e6 --with-ethernet`
--> Bitstream generated successfully
--> BIOS running on Hardware successfully. All test cases passed including SDRAM. 
Generation of demo.bin (baremetal)
` litex_bare_metal_demo --build-path=./`
Issues faced

1. Relocation truncated error 
```
 lalit@fpga-ubuntu-server:~/Pictures/work_Linux_Litex_Rocket/build/xilinx_ac701$ litex_bare_metal_demo --build-path=./
 cp: -r not specified; omitting directory '/home/lalit/Pictures/work_Linux_Litex_Rocket/litex/litex/soc/software/demo/__pycache__'
 CC       isr.o
 CC       donut.o
 CC       helloc.o
 CC       main.o
 CC       crt0.o
 CC       demo.elf
 /tmp/ccefcKZY.ltrans0.ltrans.o: in function `.L0 ':
 /home/lalit/Pictures/work_Linux_Litex_Rocket/build/xilinx_ac701/demo/isr.c:48:(.text+0x108): relocation truncated to fit: R_RISCV_HI20 against `.LC1'
 /tmp/ccefcKZY.ltrans0.ltrans.o: in function `.L0 ':
 /home/lalit/Pictures/work_Linux_Litex_Rocket/build/xilinx_ac701/demo/main.c:202:(.text.startup+0x36): relocation truncated to fit: R_RISCV_HI20 against `.LC0'
 .././/software/libc/libc.a(libc_tinystdio_printf.c.o): in function `printf':
 /home/lalit/Pictures/work_Linux_Litex_Rocket/build/xilinx_ac701/software/libc/../../../../pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/tinystdio/printf.c:42:(.text.printf+0x4): relocation truncated to fit: R_RISCV_HI20 against symbol `stdout' defined in .srodata section in /tmp/ccefcKZY.ltrans0.ltrans.o
 .././/software/libc/libc.a(libc_tinystdio_vfiprintf.c.o): in function `vfprintf':
 /home/lalit/Pictures/work_Linux_Litex_Rocket/build/xilinx_ac701/software/libc/../../../../pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/tinystdio/vfprintf.c:334:(.text.vfprintf+0x1b2): relocation truncated to fit: R_RISCV_HI20 against `.LC1'
 collect2: error: ld returned 1 exit status
 make: *** [Makefile:21: demo.elf] Error 1
 cp: cannot stat 'demo/demo.bin': No such file or directory
```
 Resolved the above error by adding the following flag [here](https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/rocket/core.py#L120), after line number: 120
` flags += "-mcmodel=medany"`

2. undefined referece to plic_init
```
 /opt/riscv_litex/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: crt0.o: in function `.L0 ':
 /home/lalit/Pictures/work_Linux_Litex_Rocket/litex/litex/soc/cores/cpu/rocket/crt0.S:101: undefined reference to `plic_init'
 collect2: error: ld returned 1 exit status
 make: *** [Makefile:21: demo.elf] Error 1
 cp: cannot stat 'demo/demo.bin': No such file or directory		
```
 The above error was resolved by replacing the isr.c file from [https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/) to [https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/demo/](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/demo/)

After the above 2 issues were resolved, demo.bin was generated.
Loading demo.bin via lxterm via serial port to AC701 hardware
`sudo lxterm /dev/ttyUSB1 --kernel demo.bin --kernel-adr 0x80000000
`
Logs
```

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec 28 2021 10:36:56
 BIOS CRC passed (fec48173)

 Migen git sha1: 9a0be7a
 LiteX git sha1: 9482cbc8

--=============== SoC ==================--
CPU:		RocketRV64[imac] @ 50MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
SDRAM:		1048576KiB 32-bit @ 400MT/s (CL-7 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 m2:0 m3:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |11111111111111000000000000000000| delays: 07+-07
  m0, b02: |00000000000000000011111111111111| delays: 25+-07
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 07+-07
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |11111111111110000000000000000000| delays: 06+-06
  m1, b02: |00000000000000000011111111111111| delays: 25+-07
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: 25+-07
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |11111111111100000000000000000000| delays: 06+-06
  m2, b02: |00000000000000001111111111111111| delays: 24+-08
  m2, b03: |00000000000000000000000000000000| delays: -
  m2, b04: |00000000000000000000000000000000| delays: -
  m2, b05: |00000000000000000000000000000000| delays: -
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b02 delays: 24+-08
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |11111111111111000000000000000000| delays: 06+-06
  m3, b02: |00000000000000000011111111111111| delays: 25+-07
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00000000000000000000000000000000| delays: -
  m3, b05: |00000000000000000000000000000000| delays: -
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b02 delays: 24+-07
Switching SDRAM to hardware control.
Memtest at 0x80000000 (2.0MiB)...
  Write: 0x80000000-0x80200000 2.0MiB     
   Read: 0x80000000-0x80200000 2.0MiB     
Memtest OK
Memspeed at 0x80000000 (Sequential, 2.0MiB)...
  Write speed: 10.8MiB/s
   Read speed: 15.5MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading demo.bin to 0x80000000 (5836 bytes)...
[LXTERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LXTERM] Upload complete (9.9KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
Executing booted program at 0x80000000

--============= Liftoff! ===============--
```

After Liftoff, unable to get any output from demo.bin 
Kindly help to look into this issue.
Note: The demo.bin works perfectly fine in the same hardware for vexriscv cpu. Issue is faced only while using rocket cpu. 
Thanks in advance

developandplay, 2022-03-18:
Removing `-flto` [here](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/common.mak#L56) seems to be the intermediate fix for the demo to work on Rocket.

developandplay, 2022-03-20:
While this does work it also breaks some other stuff.
The real issue is that the version of the toolchain that you and I were using is broken.
Update your toolchain to a more recent on like [this](https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2022.03.09) and it should work again.

Aravindh-Swaminathan, 2022-03-30:
Hi @developandplay ,
I tried your solutions in the following order. 
1) Updated the riscv-gnu-toolchain to the latest one (for ubuntu 20.04 version) and checked. The demo application code did not run after Liftoff
2) Removed the -flto flag from common.mak file and tried again. The demo application code runs perfectly fine. 

From this we can say riscv-gnu-toolchain might not be the issue. Kindly advise. 

Thanks.

developandplay, 2022-03-30:
Did you compile both the BIOS and the demo application code with the new toolchain?
For me it would only work if both of them use the latest toolchain.

Aravindh-Swaminathan, 2022-03-30:
@developandplay Yes! I did compile both BIOS and demo application using latest toolchain only since I did not want to leave anything to chance. Still did not work .

Aravindh-Swaminathan, 2022-04-01:
Hi @developandplay ,
I have reverified from my end. 
- Made sure to use latest toolchain
- added -flto flag in the common.mak file 
- compiled both BIOS and demo.bin - **Application did not run after Liftoff**
- subsequently removed the -flto flag and recompiled BIOS and demo.bin - **Application works successfully after Liftoff without any issues**

developandplay, 2022-04-01:
Just checked again myself. Sorry, I guess I was the one using a `demo.bin` without `-flto`.
I just edited the `Makefile` for the `demo.bin` to not use `-flto`.
@Aravindh-Swaminathan Could you check that https://github.com/enjoy-digital/litex/pull/1259 works for you?

@enjoy-digital Could removing `-flto` from the demo Makefile break something else?

Aravindh-Swaminathan, 2022-04-01:
Hi @developandplay ,
Checked with #1259 as per your suggestion and it works perfectly fine.
Thanks!