enjoy-digital/litex #1174: yosys+nextpnr toolchain support

suarezvictor, 2022-01-21:
Done synthesis with yosys+nextpnr on a Xilinx Artix-7 FPGA (no vivado, no symbiflow).
Tested with a FemtoRV CPU at 50MHz. It still requires manual removal of unised FD primitives intances in the generated verilog, and some small tweaking on the soc generating script (the ones giving a warning in Vivado[1]).

comand used: `./digilent_arty_basic.py --cpu-type=femtorv --build --toolchain=yosys+nextpnr`

Here's the makefile generator
```
#
# This file is part of LiteX.
#
# Copyright (c) 2020 Antmicro <www.antmicro.com>
# Copyright (c) 2020 Florent Kermarrec <florent@enjoy-digital.fr>
# Copyright (c) 2022 Victor Suarez Rovere <suarezvictor@gmail.com>
# SPDX-License-Identifier: BSD-2-Clause

import os
import subprocess
import sys
import math
from typing import NamedTuple, Union, List
import re
from shutil import which

from migen.fhdl.structure import _Fragment, wrap, Constant
from migen.fhdl.specials import Instance

from litex.build.generic_platform import *
from litex.build.xilinx.vivado import _xdc_separator, _format_xdc, _build_xdc
from litex.build import tools
from litex.build.xilinx import common

def _unwrap(value):
    return value.value if isinstance(value, Constant) else value

# Makefile -----------------------------------------------------------------------------------------

class _MakefileGenerator:
    class Var(NamedTuple):
        name: str
        value: Union[str, List[str]] = ""

    class Rule(NamedTuple):
        target: str
        prerequisites: List[str] = []
        commands: List[str] = []
        phony: bool = False

    def __init__(self, ast):
        self.ast = ast

    def generate(self):
        makefile = []
        for entry in self.ast:
            if isinstance(entry, str):
                makefile.append(entry)
            elif isinstance(entry, self.Var):
                if not entry.value:
                    makefile.append(f"{entry.name} :=")
                elif isinstance(entry.value, list):
                    indent = " " * (len(entry.name) + len(" := "))
                    line = f"{entry.name} := {entry.value[0]}"
                    for value in entry.value[1:]:
                        line += " \\"
                        makefile.append(line)
                        line = indent + value
                    makefile.append(line)
                elif isinstance(entry.value, str):
                    makefile.append(f"{entry.name} := {entry.value}")
                else:
                    raise
[... 365 lines cut ...]
    if args.with_spi_sdcard:
        soc.add_spi_sdcard()
    if args.with_sdcard:
        soc.add_sdcard()

    builder = Builder(soc, **builder_argdict(args))
    builder_kwargs = vivado_build_argdict(args) if args.toolchain == "vivado" else {}
    builder.build(**builder_kwargs, run=args.build)

    if args.load:
        prog = soc.platform.create_programmer()
        prog.load_bitstream(os.path.join(builder.gateware_dir, soc.build_name + ".bit"))

if __name__ == "__main__":
    main()
```

With yosys + nextpnr it took 10' in my machine (core i3, single core), while Vivado takes 5' using two cores.

[1] The involved flip-flops are these:

```
WARNING: [DRC PLHOLDVIO-2] Non-Optimal connections which could lead to hold violations: A LUT clk100_inst is driving clock pin of 8 cells. This could lead to large hold time violations. Involved cells are:
FD, FD_1, FD_2, FD_3, FD_4, FD_5, FD_6, and FD_7
```

gsomlo, 2022-01-21:
Interesting! What version of nextpnr did you use? I didn't notice the "upstream" version (https://github.com/yosyshq/nextpnr) advertising Xilinx support, did I miss anything? @gatecat

I'm packaging nextpnr for Fedora, and would love nothing more than to update to a version that has xilinx support :)

suarezvictor, 2022-01-22:
I use this version:
```
$ nextpnr-xilinx --version
nextpnr-xilinx -- Next Generation Place and Route (Version cd8b15db)
```

I was following this installation tutorial: https://github.com/BrunoLevy/learn-fpga/blob/master/FemtoRV/TUTORIALS/toolchain_arty.md

And made some tweaks: https://github.com/BrunoLevy/learn-fpga/issues/54

enjoy-digital, 2022-01-23:
Thanks @suarezvictor, very interesting. As we did with Symbiflow, we could use the Arty as the experimental board for it and add required tweaks to it if needed. (Once the tweaks are not longer required we could extend support to other boards). Would you mind doing a PR with your work? I could look at avoiding the manual changes you had to do. Having this in place will be a good way to enable users to test it and make fast progress all together.

suarezvictor, 2022-01-23:
Yes I'll be quite glad to prepare the PR and join efforts

suarezvictor, 2022-01-24:
See PR here, with cleanups: https://github.com/enjoy-digital/litex/pull/1180
Commits for libsdcard should be ignored :-/

suarezvictor, 2022-01-24:
RapidWright can be an option to this toolchain too, as used in the following example:
https://github.com/gatecat/nextpnr-xilinx/blob/xilinx-upstream/xilinx/examples/blinky/blinky.sh

enjoy-digital, 2022-03-29:
We can now close this since https://github.com/enjoy-digital/litex/pull/1180 has been merged. Thanks @suarezvictor.

closed 2022-03-29 by enjoy-digital