enjoy-digital/litex #1175: Adding support for other CV32-based cores

pcotret, 2022-01-22:
Hi,
I'm currently working with cores from the CV32 family. I'd like to add support for it in LiteX.
As far as I understand from the CV32E40P PR (https://github.com/enjoy-digital/litex/pull/535), I need to:
- Modify similar files
- Add configuration in `pythondata-auto`
And it will create in a way the `pythondata-cpu-cv32***` repo structure.
Am I right?

References used:
https://github.com/enjoy-digital/litex/wiki/Add-A-New-CPU

mithro, 2022-01-22:
Yes, that is correct. I will create the `pythondata-cpu-cv32***` repositories for you after you send the `pythondata-auto` patch.

pcotret, 2022-01-26:
Just to be sure, for preliminary tests before the PR, I assume that I should create the `pythondata-cpu-xxx` repo by myself.
`pythondata-auto` will only be used once I've pushed things here.

Is it correct?

mithro, 2022-01-26:
You don't need to create the `pythondata-cpu-xxx`, just send a pull request to `pythondata-auto` and I'll create the repos.

closed 2022-02-01 by pcotret