enjoy-digital/litex #1192: pythondata-cpu-vexriscv fails to build (if a modification is needed)

Acathla-fr, 2022-02-01:
Cannot build a CPU anymore :

```
fabien@XPS13:~/prog/litex/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog$ make VexRiscv_Lite.v
sbt compile "runMain vexriscv.GenCoreDefault --iCacheSize 2048 --dCacheSize 0 --mulDiv true --singleCycleShift false --singleCycleMulDiv false  --outputFile VexRiscv_Lite"
[info] Loading settings for project verilog-build from plugins.sbt ...
[info] Loading project definition from /home/fabien/prog/litex/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog/project
[info] Loading settings for project root from build.sbt ...
[info] Set current project to VexRiscvOnWishbone (in build file:/home/fabien/prog/litex/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog/)
[info] Executing in batch mode. For better performance use sbt's shell
[info] Updating ...
[warn] 	module not found: default#vexriscv_2.11;0.1.0-SNAPSHOT
[warn] ==== local: tried
[warn]   /home/fabien/.ivy2/local/default/vexriscv_2.11/0.1.0-SNAPSHOT/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/default/vexriscv_2.11/0.1.0-SNAPSHOT/vexriscv_2.11-0.1.0-SNAPSHOT.pom
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: default#vexriscv_2.11;0.1.0-SNAPSHOT: not found
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		default:vexriscv_2.11:0.1.0-SNAPSHOT
[warn] 		  +- com.github.spinalhdl:vexriscvonwishbone_2.11:0.1.0-SNAPSHOT
[error] sbt.librarymanagement.ResolveException: unresolved dependency: default#vexriscv_2.11;0.1.0-SNAPSHOT: not found
[error] 	at sbt.internal.librarymanagement.IvyActions$.resolveAndRetrieve(IvyActions.scala:332)
[error] 	at sbt.internal.librarymanagement.IvyActions$.$anonfun$updateEither$1(IvyActions.scala:208)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.$anonfun$withModule$1(Ivy.scala:239)
[error] 	at sbt.internal.librarymanagement.IvySbt.$anonfun$withIvy$1(Ivy.scala:204)
[error] 	at sbt.internal.librarymanagement.IvySbt.sbt$internal$librarymanagement$IvySbt$$action$1(Ivy.scala:70)
[error] 	at sbt.internal.librarymanagement.IvySbt$$anon$3.call(Ivy.scala:77)
[error] 	at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:113)
[error] 	at xsbt.boot.Locks$GlobalLock.withChannelRetries$1(Locks.scala:91)
[error] 	at xsbt.boot.Locks$GlobalLock.$anonfun$withFileLock$1(Locks.scala:119)
[error] 	at xsbt.boot.Using$.withResource(Using.scala:12)
[error] 	at xsbt.boot.Using$.apply(Using.scala:9)
[error] 	at xsbt.boot.Locks$GlobalLock.withFileLock(Locks.scala:119)
[error] 	at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:71)
[error] 	at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:59)
[error] 	at xsbt.boot.Locks$.apply0(Locks.scala:47)
[error] 	at xsbt.boot.Locks$.apply(Locks.scala:36)
[error] 	at sbt.internal.librarymanagement.IvySbt.withDefaultLogger(Ivy.scala:77)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:199)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:196)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.withModule(Ivy.scala:238)
[error] 	at sbt.internal.librarymanagement.IvyActions$.updateEither(IvyActions.scala:193)
[error] 	at sbt.librarymanagement.ivy.IvyDependencyResolution.update(IvyDependencyResolution.scala:20)
[error] 	at sbt.librarymanagement.DependencyResolution.update(DependencyResolution.scala:56)
[error] 	at sbt.internal.LibraryManagement$.resolve$1(LibraryManagement.scala:45)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$12(LibraryManagement.scala:93)
[error] 	at sbt.util.Tracked$.$anonfun$lastOutput$1(Tracked.scala:68)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$19(LibraryManagement.scala:106)
[error] 	at scala.util.control.Exception$Catch.apply(Exception.scala:224)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11(LibraryManagement.scala:106)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11$adapted(LibraryManagement.scala:89)
[error] 	at sbt.util.Tracked$.$anonfun$inputChanged$1(Tracked.scala:149)
[error] 	at sbt.internal.LibraryManagement$.cachedUpdate(LibraryManagement.scala:120)
[error] 	at sbt.Classpaths$.$anonfun$updateTask$5(Defaults.scala:2561)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:44)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:40)
[error] 	at sbt.std.Transform$$anon$4.work(System.scala:67)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:269)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:16)
[error] 	at sbt.Execute.work(Execute.scala:278)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:269)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:178)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:37)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
[error] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error] 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error] 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error] 	at java.lang.Thread.run(Thread.java:748)
[error] (update) sbt.librarymanagement.ResolveException: unresolved dependency: default#vexriscv_2.11;0.1.0-SNAPSHOT: not found
[error] Total time: 1 s, completed Feb 1, 2022 6:31:40 PM
make: *** [Makefile:12: VexRiscv_Lite.v] Error 1

```

mithro, 2022-02-01:
@tcal-x - Any idea?

tcal-x, 2022-02-01:
@Acathla-fr did you load the submodules for pythondata-cpu-vexriscv?   It needs to load the VexRiscv source to do a build.  It will be loaded at ext/VexRiscv/ relative to the directory where you do the `make`.   From where you are, just do `git submodule update --init`.

If you *have* already loaded that submodule and still get this error, then there may be a Scala issue, and I'd ask @Dolu1990 in that case.

tcal-x, 2022-02-01:
@Acathla-fr , I don't think it's documented well, if at all.   

I'm considering adding a check and useful message in the Makefile...

Acathla-fr, 2022-02-02:
I had to clear the folder for git to be happy.
The submodule takes only 28MB of disk space in ext/, maybe it can be downloaded automatically...

Dolu1990, 2022-02-04:
@Acathla-fr How did you got the repository on your harddrive ? did you cloned it manualy or you used some scripts to setup things ?

Acathla-fr, 2022-02-04:
@Dolu1990 I used litex's setup script : https://github.com/enjoy-digital/litex/blob/master/litex_setup.py

Dolu1990, 2022-02-07:
Ahhh, i was confused between the smp and non-smp vexriscv repo.

Got https://github.com/litex-hub/pythondata-cpu-vexriscv/pull/17 merged

So the reason why Vex repo isn't cloned recursively is to avoid bloating things. 28 MB isn't much by itself, but as the Litex "depedencies" grows, it contribute to the keep things not too big.

enjoy-digital, 2022-03-29:
A check has been added in the Makefile of pythondata-cpu-vexriscv. We can now close this.

closed 2022-03-29 by enjoy-digital