enjoy-digital/litex #1194: I2C automatic initialization : Fix it when no "i2c" module present in design.

enjoy-digital, 2022-02-02:
I2C automatic initialization is possible on multiple I2C interfaces with arbitrary names but one interface has to be named `i2c` for it to properly work. We should avoid this since time can be lost understanding the issue when interfaces are named differently.

geertu, 2022-02-02:
For i2c0, see 347549797a5c3fd34a913182d6d4c118b46d22e3

enjoy-digital, 2022-02-02:
Thanks @geertu, after this we indeed want to try to simplify handling of multiple similar peripherals by the BIOS and fix numbering. Sorry if it's a bit slow.

fjullien, 2022-02-03:
This is what I've done: https://github.com/enjoy-digital/litex/pull/1195
Now arbitrary i2c module names can be used and we can now choose the active i2c bus in the BIOS.

enjoy-digital, 2022-02-04:
Thanks @fjullien, we can close this. We'll probably polish this over time but for now it's enough for the targeted use case.

closed 2022-02-04 by enjoy-digital