enjoy-digital/litex #1198: 16-bit mode video: SoC builder fails

suarezvictor, 2022-02-04:
When using "rgb565" the SoC builder fails

Example code for BaseSoC:
```
        if with_video_framebuffer:
            from litex.soc.cores.video import VideoVGAPHY
            self.submodules.videophy = VideoVGAPHY(platform.request("vga"), clock_domain="vga")
            self.add_video_framebuffer(phy=self.videophy, timings="800x600@60Hz", clock_domain="vga", format="rgb565")
```
Produces the following error (only if format is set to "rgb565"):
```
Traceback (most recent call last):
  File "./digilent_arty_video.py", line 210, in <module>
    main()
  File "./digilent_arty_video.py", line 203, in main
    builder.build(**builder_kwargs, run=args.build)
  File "/home/vsuarez/litex/litex/soc/integration/builder.py", line 336, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/vsuarez/litex/litex/soc/integration/soc.py", line 1145, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/vsuarez/litex/litex/build/xilinx/platform.py", line 65, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/vsuarez/litex/litex/build/xilinx/vivado.py", line 349, in build
    v_output = platform.get_verilog(fragment, name=build_name, **kwargs)
  File "/home/vsuarez/litex/litex/build/xilinx/platform.py", line 59, in get_verilog
    **kwargs)
  File "/home/vsuarez/litex/litex/build/generic_platform.py", line 420, in get_verilog
    return verilog.convert(fragment, platform=self, **kwargs)
  File "/home/vsuarez/litex/litex/gen/fhdl/verilog.py", line 602, in convert
    verilog += _print_combinatorial_logic_synth(f, ns)
  File "/home/vsuarez/litex/litex/gen/fhdl/verilog.py", line 471, in _print_combinatorial_logic_synth
    r += _print_node(ns, _AT_NONBLOCKING, 1, g[1])
  File "/home/vsuarez/litex/litex/gen/fhdl/verilog.py", line 287, in _print_node
    return "".join(_print_node(ns, at, level, n, target_filter) for n in node)
  File "/home/vsuarez/litex/litex/gen/fhdl/verilog.py", line 287, in <genexpr>
    return "".join(_print_node(ns, at, level, n, target_filter) for n in node)
  File "/home/vsuarez/litex/litex/gen/fhdl/verilog.py", line 292, in _print_node
    r += _print_node(ns, at, level + 1, node.t, target_filter)
  File "/home/vsuarez/litex/litex/gen/fhdl/verilog.py", line 287, in _print_node
    return "".join(_print_node(ns, at, level, n, target_filter) for n in node)
  File "/home/vsuarez/litex/litex/gen/fhdl/verilog.py", line 287, in <genexpr>
    return "".join(_print_node(ns, at, level, n, target_filter) for n in node)
  File "/home/vsuarez/litex/litex/gen/fhdl/verilog.py", line 283, in _print_node
    return "\t"*level + _print_expression(ns, node.l)[0] + assignment + _print_expression(ns, node.r)[0] + ";\n"
  File "/home/vsuarez/litex/litex/gen/fhdl/verilog.py", line 249, in _print_expression
    return _print_slice(ns, node)
  File "/home/vsuarez/litex/litex/gen/fhdl/verilog.py", line 212, in _print_slice
    assert (node.stop - node.start) >= 1
AssertionError

```

enjoy-digital, 2022-03-29:
@suarezvictor: Can you provide the adapations done to be able to reproduce?

suarezvictor, 2022-03-29:
yes, this is the full script tested with latest version of litex
It works with "rgb888" but not with "rgb565"

```
#!/usr/bin/env python3

#
# This file is part of LiteX-Boards.
#
# Copyright (c) 2015-2019 Florent Kermarrec <florent@enjoy-digital.fr>
# Copyright (c) 2020 Antmicro <www.antmicro.com>
# SPDX-License-Identifier: BSD-2-Clause

import os
import argparse

from migen import *

from litex_boards.platforms import arty
from litex.build.xilinx.vivado import vivado_build_args, vivado_build_argdict

from litex.soc.cores.clock import *
from litex.soc.integration.soc import SoCRegion
from litex.soc.integration.soc_core import *
from litex.soc.integration.builder import *
from litex.soc.cores.led import LedChaser
from litex.soc.cores.gpio import GPIOTristate

from litedram.modules import MT41K128M16
from litedram.phy import s7ddrphy

from liteeth.phy.mii import LiteEthPHYMII

from litespi.modules import S25FL128L
from litespi.opcodes import SpiNorFlashOpCodes as Codes
from litespi.phy.generic import LiteSPIPHY
from litespi import LiteSPI

"""
##https://github.com/projf/projf-explore/blob/master/graphics/fpga-graphics/xc7/arty.xdc
## Project F: FPGA Graphics - Arty A7-35T Board Constraints
## (C)2021 Will Green, open source hardware released under the MIT License
## VGA Pmod on Header JB/JC
##("pmodb", "E15 E16 D15 C15 J17 J18 K15 J15"),
##("pmodc", "U12 V12 V10 V11 U14 V14 T13 U13"),

set_property -dict {PACKAGE_PIN E15 IOSTANDARD LVCMOS33} [get_ports {vga_r[0]}];
set_property -dict {PACKAGE_PIN E16 IOSTANDARD LVCMOS33} [get_ports {vga_r[1]}];
set_property -dict {PACKAGE_PIN D15 IOSTANDARD LVCMOS33} [get_ports {vga_r[2]}];
set_property -dict {PACKAGE_PIN C15 IOSTANDARD LVCMOS33} [get_ports {vga_r[3]}];
set_property -dict {PACKAGE_PIN J17 IOSTANDARD LVCMOS33} [get_ports {vga_b[0]}];
set_property -dict {PACKAGE_PIN J18 IOSTANDARD LVCMOS33} [get_ports {vga_b[1]}];
set_property -dict {PACKAGE_PIN K15 IOSTANDARD LVCMOS33} [get_ports {vga_b[2]}];
set_property -dict {PACKAGE_PIN J15 IOSTANDARD LVCMOS33} [get_ports {vga_b[3]}];

set_property -dict {PACKAGE_PIN U12 IOSTANDARD LVCMOS33} [get_ports {vga_g[0]}];
set_property -dict {PACKAGE_PIN V12 IOSTANDARD LVCMOS33} [get_ports {vga_g[1]}];
set_property -dict {PACKAGE_PIN V10 IOSTANDARD LVCMOS33} [get_ports {vga_g[2]}];
set_property -dict {PACKAGE_PIN V11 IOSTANDARD LVCMOS33} [get_ports {vga_g[3]}];
set_property -dict {PACKAGE_PIN U14 IOSTANDARD LVCMOS33} [get_ports {vga_hsync}];
set_property -dict {PACKAGE_PIN V14 IOSTANDARD LVCMOS33} [get_ports {vga_vsync}];
"""
# CRG ----------------------------------------------------------------------------------------------

class _CRG(Module):
    def __init__(self, platform, sys_clk_freq, with_rst=True):
        self.rst = Signal()
        self.clock_domains.cd_sys       = ClockDomain()
        self.clock_domains.cd_sys4x     = ClockDomain(reset_less=True)
        self.clock_domains.cd_sys4x_dqs = ClockDomain(reset_less=True)
[... 172 lines cut ...]
        with_jtagbone  = args.with_jtagbone,
        with_mapped_flash = args.with_mapped_flash,
        with_spi_flash = args.with_spi_flash,
        with_pmod_gpio = args.with_pmod_gpio,
        **soc_core_argdict(args)
    )
    if args.sdcard_adapter == "numato":
        soc.platform.add_extension(arty._numato_sdcard_pmod_io)
    else:
        soc.platform.add_extension(arty._sdcard_pmod_io)
    if args.with_spi_sdcard:
        soc.add_spi_sdcard()
    if args.with_sdcard:
        soc.add_sdcard()
    builder = Builder(soc, **builder_argdict(args))
    builder_kwargs = vivado_build_argdict(args) if args.toolchain == "vivado" else {}
    builder.build(**builder_kwargs, run=args.build)

    if args.load:
        prog = soc.platform.create_programmer()
        prog.load_bitstream(os.path.join(builder.gateware_dir, soc.build_name + ".bit"))

if __name__ == "__main__":
    main()
```

enjoy-digital, 2022-04-22:
@suarezvictor: Thanks for reporting and sorry for the delay, this is fixed with https://github.com/enjoy-digital/litex/issues/1198. The issue was that the depth checks were done in FPGA logic, not in Python and this was generating invalid logic. I haven't noticed this when merging the 16-bit support PR. This should now be fine (at least compiles fine).

closed 2022-04-22 by enjoy-digital