enjoy-digital/litex #1200: cv32e41p - Prompt not answering

pcotret, 2022-02-06:
Still working on porting the [cv32e41p](https://github.com/openhwgroup/cv32e41p). LiteX boots but the prompt doesn't answering...
 
[image]

- As everything boots, I assume hardware connections to the core are OK.
- The prompt blinks but doesn't print anything from my keyboard.

What could be the problem?
Something related to interruptions maybe?  (`litex/bios/isr` has been modified).

mithro, 2022-02-07:
My guess would be interrupts too...

enjoy-digital, 2022-02-08:
@pcotret: To confirm it's related to interrupts, can you add `self.add_constant("UART_POLLING")` to your SoC? It will switch the UART to polling mode (as we are doing for CPUs not supporting interrupts: https://github.com/enjoy-digital/litex/blob/8f5d9714917cedc7c39b440357380cd4463b9b4d/litex/soc/integration/soc.py#L1253).

pcotret, 2022-02-08:
As I'm using the `litex_sim` binary, I've added a flag in my `core/cv32e41p/core.py` : 
```diff
    # GCC Flags.
    @property
    def gcc_flags(self):
        flags = GCC_FLAGS[self.variant]
        flags += "-D__cv32e41p__ "
+       flags += f" -DUART_POLLING"
 ```
[image]

So, if I understand correctly, guilty part is related to interrupts?

enjoy-digital, 2022-02-08:
@pcotret: If it's working correctly when defining `UART_POLLING` the issue is indeed related to interrupts.

pcotret, 2022-02-11:
[image]

Found my issue!
In fact, the UART interrupt was mapped on read-only bits in related CSRs...

closed 2022-02-11 by pcotret