enjoy-digital/litex #1202: lab004 makefile error: unrecognized emulation mode

marrrk, 2022-02-07:
hello, i've been working through the labs and i've encountered an error on lab004 that i'm not quite sure how to fix. when running the commands in step 2, i get the following error:

```
$ make all
 CC       main.o
 LD       firmware.elf
riscv64-unknown-elf-ld: unrecognised emulation mode: abi=ilp32
Supported emulations: elf64lriscv elf32lriscv
make: *** [Makefile:18: firmware.elf] Error 1
```

i've ensured that i'm using the compiler installed in `litex_setup.py` by running it with the argument `--gcc=riscv` and adding that bin folder produced to my path.

any help is greatly appreciated!

Esteves4, 2022-03-19:
Hello,

me, @alexandrekj and @MCassidy2411 were having the same problem. We solved by changing a few things in the firmware/Makefile file.

Begining in line 17:

- we changed LD to CC
- the crt0.o is in the bios directory
- The Libc is also needed by the c-code
- The Libbase changed its name from -lbase-nofloat to -lbase

```
firmware.elf: $(OBJECTS)
	$(CC) $(LDFLAGS) \
		-T linker.ld \
		-N -o $@ \
		 $(BUILD_DIR)/software/bios/crt0.o \
		$(OBJECTS) \
		-L$(BUILD_DIR)/software/libbase \
		-L$(BUILD_DIR)/software/libcompiler_rt \
		-L$(BUILD_DIR)/software/libc \
		-lbase -lcompiler_rt -lc
	chmod -x $@
```

marrrk, 2022-03-22:
Hi, thanks for commenting!
I too noticed a few of these things but was unable to fix it 100%. One solution I was exploring was compiling the crt0.S found in `$(CPU_DIRECTORY)` and using directly but that led to a different error. I am yet to deduce whether it is caused by the change or something else wrong with the toolchain.

Trying your fix, I discovered something. For some reason my copy of litex deletes the crt0.o file in the bios directory and so I get a file not found error. I tried building the SoC again and noticed the following output during the build process:
```
make: Entering directory '/home/markn/riscv/code/fpga_101-master/lab004/build/software/bios'
 CC       boot.o
 CC       cmd_bios.o
 CC       cmd_mem.o
 CC       cmd_boot.o
 CC       cmd_i2c.o
 CC       cmd_spiflash.o
 CC       cmd_litedram.o
 CC       cmd_liteeth.o
 CC       cmd_litesdcard.o
 CC       cmd_litesata.o
 CC       sim_debug.o
 CC       main.o
 CC       crt0.o
 CC       bios.elf
chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m litex.soc.software.mkmscimg bios.bin --little
python3 -m litex.soc.software.memusage bios.elf /home/markn/riscv/code/fpga_101-master/lab004/build/software/bios/../include/generated/regions.ld riscv64-unknown-elf

ROM usage: 21.07KiB 	(65.83%)
RAM usage: 1.60KiB 	(20.02%)

rm crt0.o
```
Updating Litex doesn't change this.
Currently, I'm looking for where to find/fix that.

closed 2022-03-28 by enjoy-digital