enjoy-digital/litex #1205: mem_test giving error in QSPI/ SPI Flash memory region

maninigaur, 2022-02-10:
I am trying to integrate SPI flash functionality on Xilinx AC701 board Litex with VexRISCV CPU. After making some changes in the `litex-boards/targets/xilinx_ac701.py`, `litex-boards/platforms/xilinx_ac701.py` and` litex/soc/integration/soc.py` (modifications pasted below), SPI Flash is initialized in the BIOS and memory region allocated. However, when running mem_test, it shows "memtest access error" and exits. Similarly, the data written on some SPI Flash location using mem_write command is not reproduced with mem_read. 
The attempt here is to validate the working of SPI Flash on BIOS before trying to flash a boot or firmware image. 

Additions made in the code to integrate SPI Flash (N25Q256A13ESF40G- Micron) on Xilinx AC701:
[targets/ac701.py](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/xilinx_ac701.py#L134)
```
# SPI Flash --------------------------------------------------------------------------------
        if with_spi_flash:
            from litespi.modules import  N25Q256A
            from litespi.opcodes import SpiNorFlashOpCodes as Codes
            self.add_spi_flash(mode="4x", module= N25Q256A(Codes.READ_1_1_4), with_master=True)
```
[platforms/ac701.py](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/xilinx_ac701.py#L136)
 ```
# SPIFlash
    ("spiflash4x", 0,
        Subsignal("cs_n", Pins("P18")),
        Subsignal("dq",   Pins("R14 R15 P14 N14")),
        IOStandard("LVCMOS33")
    ),
```
[soc/integration/soc.py](https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc.py#L1569)
`self.add_constant("FLASH_BOOT_ADDRESS", 0x02000000)`

SPI Flash Logs from BIOS and memory regions with mem_list:
```
Initializing N25Q256A SPI Flash @0x02000000...
Enabling Quad mode...
SPI Flash clk configured to 50 MHz
Memspeed at 0x2000000 (Sequential, 4.0KiB)...
   Read speed: 13.0MiB/s
Memspeed at 0x2000000 (Random, 4.0KiB)...
   Read speed: 4.2MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from flash...
Error: Invalid image length 0xffffffff
Booting from network...
Local IP: 192.168.1.50
Remote IP: 192.168.1.100
Booting from boot.json...
Booting from boot.bin...
Copying boot.bin to 0x40000000... 
Network boot failed.
No boot medium found

--============= Console ================--

litex> mem_list
Available memory regions:
ROM       0x00000000 0x20000 
SRAM      0x10000000 0x2000 
MAIN_RAM  0x40000000 0x10000000 
ETHMAC    0x80000000 0x2000 
SPIFLASH  0x02000000 0x2000000 
CSR       0xf0000000 0x10000 

```
Running mem_test, mem_write and mem_read:
```
litex> mem_test 0x2000000 0x2000000 
Memtest at 0x2000000 (32.0MiB)...
memtest_access error @ 0x2000000, exiting memtest.

litex> mem_read 0x02000000
Memory dump:
0x02000000  ff ff ff ff                                      ....            

litex> mem_write 0x02000000 0xabababab

litex> mem_read 0x02000000 0x4        
Memory dump:
0x02000000  ff ff ff ff  
```  
My questions are:
1. Do mem_read and mem_write even work on the SPI flash region?
2. Used Opcode READ_1_1_4 in this test case, but I have also tried with other opcodes available with this device module. Is it correct?
3. What am I missing here?

enjoy-digital, 2022-02-12:
Hello,

for now writing to SPI Flash is not supported by LiteSPI, so only `mem_read` can be used, not mem_test`. To verify that the core is behaving correctly and that the integration is correct, you can flash some specific contents to the Flash (with OpenOCD for example) and verify with `mem_read` that the content is correctly read by `mem_read`.

Once you have this, you'll be able to boot from it. If issues, it could be useful to downgrade to `READ_1_1_1` (the QE bit could not be configured correctly on your SPI Flash) and if still not working look at the QSPI signals with a logic analyzer.

harshitk26, 2022-02-17:
hi @enjoy-digital 
for AC701 board we have to flash proxy bitfile for accessing Flash memory, but not sure if openocd (with proxy flash bit file) will work with QSPI ?? , as I can see in 
https://github.com/quartiq/bscan_spi_bitstreams/blob/master/xilinx_bscan_spi.py#L319

PIN config is SPI based (not QSPI)
and we need to add pin config here to support 

```
added in **packages**
("fbg676-2", 1): ["P18", None, "R14", "R15"],   <---- Here we configured QSPI to use in SPI mode (not sure if that works acc to https://www.micron.com/-/media/client/global/documents/products/data-sheet/nor-flash/serial-nor/n25q/n25q_256mb_1_8v.pdf PAGE 18)
....
added in **pinouts**
"xc7a200t": ("fbg676-2", 1, "LVCMOS33", Series7),
```

enjoy-digital, 2022-04-28:
@harshitk26: With OpenOCD, the SPI Flash will be flashed in SPI mode. QSPI is mostly only interesting when using the SPI from the FPGA gateware, which is supported by LiteSPI. Support has been added to the AC701 with https://github.com/litex-hub/litex-boards/commit/9914478854b83ec5750c07df995b2390977280c2.

closed 2022-04-28 by enjoy-digital