enjoy-digital/litex #1212: Rocket Chip boot issue with boot.bin on AC701

harshitk26, 2022-02-17:
Hi everyone 
While everything is working fine with vexriscv architecture, we are able to boot bare-metal, linux and RTOS without any issue (with some few changes in dts and core py files and DDR3 issues) same we are not getting any out after liftoff on **RocketChip architecture.**

Is there any open issue with booting with Rocket Chip ? 

Generation of bit stream.
litex-boards/litex_boards/targets/xilinx_ac701.py --build --cpu-type rocket --cpu-variant linux4 --sys-clk-freq 50e6 --with-ethernet
--> Bitstream generated successfully
--> BIOS running on Hardware successfully. All test cases passed including SDRAM.
Generation of demo.bin (baremetal)
litex_bare_metal_demo --build-path=./

Loading demo.bin via lxterm via serial port to AC701 hardware
sudo lxterm /dev/ttyUSB1 --kernel demo.bin --kernel-adr 0x80000000

```

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec 28 2021 10:36:56
 BIOS CRC passed (fec48173)

 Migen git sha1: 9a0be7a
 LiteX git sha1: 9482cbc8

--=============== SoC ==================--
CPU:		RocketRV64[imac] @ 50MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
SDRAM:		1048576KiB 32-bit @ 400MT/s (CL-7 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 m2:0 m3:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |11111111111111000000000000000000| delays: 07+-07
  m0, b02: |00000000000000000011111111111111| delays: 25+-07
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 07+-07
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |11111111111110000000000000000000| delays: 06+-06
  m1, b02: |00000000000000000011111111111111| delays: 25+-07
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: 25+-07
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |11111111111100000000000000000000| delays: 06+-06
  m2, b02: |00000000000000001111111111111111| delays: 24+-08
  m2, b03: |00000000000000000000000000000000| delays: -
  m2, b04: |00000000000000000000000000000000| delays: -
  m2, b05: |00000000000000000000000000000000| delays: -
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b02 delays: 24+-08
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |11111111111111000000000000000000| delays: 06+-06
  m3, b02: |00000000000000000011111111111111| delays: 25+-07
  m3, b03: |00000000000000000000000000000000| delays: -
  m3, b04: |00000000000000000000000000000000| delays: -
  m3, b05: |00000000000000000000000000000000| delays: -
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b02 delays: 24+-07
Switching SDRAM to hardware control.
Memtest at 0x80000000 (2.0MiB)...
  Write: 0x80000000-0x80200000 2.0MiB     
   Read: 0x80000000-0x80200000 2.0MiB     
Memtest OK
Memspeed at 0x80000000 (Sequential, 2.0MiB)...
  Write speed: 10.8MiB/s
   Read speed: 15.5MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LXTERM] Received firmware download request from the device.
[LXTERM] Uploading demo.bin to 0x80000000 (5836 bytes)...
[LXTERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LXTERM] Upload complete (9.9KB/s).
[LXTERM] Booting the device.
[LXTERM] Done.
Executing booted program at 0x80000000

--============= Liftoff! ===============--
```

After Liftoff, unable to get any output from demo.bin, We are stuck here :(

developandplay, 2022-03-18:
Removing `-flto` [here](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/common.mak#L56) seems to be the intermediate fix for the demo to work on Rocket.

developandplay, 2022-03-20:
While this does work it also breaks some other stuff.
The real issue is that the version of the toolchain that you and I were using is broken.
Update your toolchain to a more recent on like [this](https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2022.03.09) and it should work again.

enjoy-digital, 2022-03-29:
Thanks @developandplay, I think we can now close this. @harshitk26: Please re-open if you still have troubles.

closed 2022-03-29 by enjoy-digital