enjoy-digital/litex #1217: A question about last changes on video core

radical3dd, 2022-02-23:
First of all I want to thank everyone for Litex. I started experimenting with it and I love it.
I want to understand the changes introduced by this commit #[25152fe](https://github.com/enjoy-digital/litex/commit/25152fe4de245f7030e03d53d26d8bdddd7079f1)

Sorry if these questions are already answered somewhere else.

The commit message said these changes was introduced to align hsync and vsync.
Why is it needed?

Why starting the vertical sync with the horizontal sync? As I understand the vertical sync should be started when `source.vcount == vsync_start`.

As I understand in Migen the last statement which update a signal wins
so does that line https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/video.py#L250
unnecessary since this other line https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/video.py#L257
will re-affect the value of the `source.vcount` and those two lines are triggered by the same condition.

This line https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/video.py#L254 said `End of HScan`
but the if check for `hsync_start` not the end of the horizontal scan.
Why this differences? which is incorrect? the comment? the check? or all of this are fine?

Thank you for your work and sorry for the disturbing if all this things are correct

fjullien, 2022-02-24:
Hi, you're right, it's a bug. I have a fix I didn't commit yet:

```
@@ -246,8 +246,9 @@ class VideoTimingGenerator(Module, AutoCSR):
                 If(source.hcount == hres,          NextValue(hactive,       0)), # End of HActive.
                 If(source.hcount == hsync_start,
                     NextValue(source.hsync, 1),                                  # Start of HSync.
-                    NextValue(source.vsync, 1),                                  # Start of VSync.
-                    NextValue(source.vcount, vsync_start)
+                    If(source.vcount == vsync_start,
+                        NextValue(source.vsync, 1)                               # Start of VSync.
+                    )
```

And about the comment that says "End of HScan", it's not true anymore.

The code is not clear as it used to be but I didn't want to change the parameters of the core.

fjullien, 2022-02-24:
At the end, we just needed to change https://github.com/enjoy-digital/litex/blob/15045e96f08915481c0684958e50c7111f91d55c/litex/soc/cores/video.py#L250 to `If(source.hcount == hsync_start` . I'll do a PR soon, thanks for your review.

enjoy-digital, 2022-02-25:
@radical3dd: This should be clearer with https://github.com/enjoy-digital/litex/pull/1227. Thanks @fjullien.

closed 2022-02-25 by enjoy-digital