enjoy-digital/litex #1220: USSystemMonitor doesn't work on Zynq Ultrascale+

smunaut, 2022-02-24:
So there are two issues :
 * It uses `SYSMONE1` while the UltrascalePlus use `SYSMONE4`  (and the formula to compute temp is a bit different, so help text is wrong).  This isn't too much of an issue because it gets upgraded automatically by vivado
 * when `SIM_DEVICE` isn't set, it sets it to default value of ULTRASCALE_PLUS  (AFAICT) and this creates a sim mismatch and somehow vivado considers that a DRC error and aborts and refuses to create a bitstream.

To fix it ATM I changed `SYSMONE1` to `SYSMONE4` and added `p_SIM_DEVICE = "ZYNQ_ULTRASCALE",` and that makes it happy.

There are several ways to handle that in the code, just no sure what way you'd prefer.
Making ` USSystemMonitor`, `USPSystemMonitor` and a `ZynqUSPSystemMonitor` all deriving from some baseclass and setting the instance name and `p_SIMDEVICE` appropriately ?   Or a constructor param ? or ...

enjoy-digital, 2022-02-24:
Thanks @smunaut,

> Making USSystemMonitor, USPSystemMonitor and a ZynqUSPSystemMonitor all deriving from some baseclass and setting the instance name and p_SIMDEVICE appropriately ?

This looks fine yes.

> Also technically VCCPSINTLP / VCCPSINTFP / VCCPSAUX only exist in the Zynq UlstraScale Plus case AFAIK. So should they be present only in that subclass ?

This should be only be present only for the devices having it physically. This could probably be enabled/disabled with the same parameter used to specify the `p_SIMDEVICE`. Happy to have a look at this if you have troubles figuring out how to do this.

smunaut, 2022-02-25:
Fixed by 8f00b0f182945f8519b40afe34d094e996ab264a
( forgot to put that in the commit message :sweat_smile: )