enjoy-digital/litex #1233: JTAGBone not operating properly on ZynqMP UltraScale Plus

smunaut, 2022-03-07:
On the `adi_adrv2crr_fmc.py` target, `JTAGBone` is present by default. I also added `UARTBone` and switched the uart to crossover.

Accesses through the `UARTBone` work fine and I can read the board identify, or access registers and all works fine.
Accesses through the `JTAGBone` just return random value. Writes also don't seem to go through.

```
$ litex_cli --ident
þÿ¿ß¿ï÷ýþß÷ûþÿ¿ß÷ûþÿ¿ß÷ûþÿ¿ß÷ûþÿ¿ß÷ûþÿ¿ßïûýÿßïûýÿ¿ï÷ýþ¿ï÷ýþ¿ï÷ýþÿ¿ß÷ûþÿ¿ßïûýÿßïûýþ¿ï÷ûþÿ¿ß÷ûþ¿ï÷ýþ¿ï÷ýþ¿ß÷ûþÿßïûýþ¿ß÷ûþÿßïûýþ¿ï÷ûþÿßï÷ýþ¿ß÷ûýÿ¿ï÷ûþÿ¿ßïûýÿßïûýþ¿ï÷ûþÿßïûýÿ¿ï÷ýþ¿ï÷ûþÿßï÷ýþÿ¿ßïûýÿ¿ï÷ûþÿ¿ß÷ûþÿ¿ßïûýÿ¿ß÷ûþÿßï÷ýþÿ¿ß÷ûý
```

```
$ litex_server --jtag --jtag-config prog/openocd_zynqmp_ft232.cfg
Open On-Chip Debugger 0.11.0-rc2
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
fpga_program
jtagstream_serve
Info : clock speed 25000 kHz
Info : JTAG tap: uscale.ps tap/device found: 0x5ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x5)
Info : JTAG tap: uscale.tap tap/device found: 0x04740093 (mfg: 0x049 (Xilinx), part: 0x4740, ver: 0x0)
Info : JTAG tap: uscale.ps tap/device found: 0x5ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x5)
Info : JTAG tap: uscale.tap tap/device found: 0x04740093 (mfg: 0x049 (Xilinx), part: 0x4740, ver: 0x0)
Error: JTAG-DP STICKY ERROR
Warn : target uscale.a53.0 examination failed
Info : starting gdb server for uscale.a53.0 on 3333
Info : Listening on port 3333 for gdb connections
[CommUART] port: JTAG / tcp port: 1234
Connected with 127.0.0.1:45506
Disconnect
```

`openocd` config : 
( The `xilinx_zynqmp.cfg` is taken from upstream but modified so that `$_CHIPNAME.tap` is the logic tap and not the ARM cores one. )

```
source [find interface/ftdi/digilent_jtag_hs3.cfg]
ftdi_tdo_sample_edge falling
adapter speed 25000

source [find cpld/xilinx_zynqmp.cfg]
source [find cpld/jtagspi.cfg]

proc fpga_program {} {
    global _CHIPNAME
    xc7_program $_CHIPNAME.tap
}
```

smunaut, 2022-03-08:
So ... first issue is that the `USER1` ... `USER4` commands have a different `IR` values. Respectively `0x902`, `0x903`, `0x922`, `0x923` according to the BSDL.

Then there is the issue that the PL tap is not alone in the chain.

The only way to properly handle arbitrary other devices in the chain is to stick to JTAG spec/semantics and that means that you capture the value you want to send in the `DR CAPTURE` state, shift TDI in your 10 bit DR and output to TDO in `DR SHIFT` mode (you'll get more than 10 shifts), then when moved to `DR UPDATE` mode, read whatever is currently in the shift register and use that as your value.

I implemented that in hardware, pretty simple logic (actually even simpler than the current one).

Problem is `openocd`.  Each exchange needs to be a distinct `drscan` command so that it goes through the `CAPTURE` / `SHIFT` / `UPDATE` states. And this is horribly slow, with massive gaps between commands ...

enjoy-digital, 2022-04-28:
Initial support based on your work @smunaut has been merged with https://github.com/enjoy-digital/litex/commit/0a738002e0d24e8a34006c4535b0e5c8a64beae3. This provides an entry point to use JTAGBone/JTAG-UART on Zynq/Zynq-MP devices but will still probably be slower than other bridge. We could have a look at speeding this up (for all devices) in the future.

closed 2022-04-28 by enjoy-digital