enjoy-digital/litex #1237: Search Engine Optimization

locriacyber, 2022-03-13:
This repository is really hard to find. I only found it through a research paper talking about Litex.

Github and other search engines only index the project name and description, along with repo tags on Github. No wonder it's hard to find.

Any plan on adding tags and making the repo description better?

enjoy-digital, 2022-03-14:
Thanks @locriacyber, I just added some tags.

closed 2022-03-14 by enjoy-digital

locriacyber, 2022-03-16:
How about project description? "Build your hardware, easily!" is a slogan, not project description.

There's probably more to be done.
If I search for "fpga softcore cpu framework site:github.com" on Google, this project is nowhere to be found.

locriacyber, 2022-03-20:
Sorry about my tone. I visited this github page before but forgot about the project name, and it took me 20+ Google searches to find this repository. I felt like it is not reachable from a search engine, and I want more people to know about the project.