enjoy-digital/litex #1238: Support of the Xilinx ZCU102 board

jwfaye, 2022-03-14:
Hello everyone, 
I am a Ph.D student working in the area of computer architecture and signal processing. For my first experimentation, I am planning to use Litex with the ZCU102 (because it is the board available in my laboratory) to up skills in computer architecture and design of systems. I need some help to integrate the ZCU102 board. I look at the scripts of the ZCU104 and ZCU106 boards. I try to understand the I/Os but  I am unable to make a relation with the doc I found here : https://www.xilinx.com/support/documentation/boards_and_kits/zcu104/ug1267-zcu104-eval-bd.pdf. I started trying to understand the pinout of the DDR4 (as it is more complex and the others should be straightforward). Can someone help me to understand that ? I will propose a script then for the ZCU102 and will will discuss it.

closed 2022-03-14 by jwfaye

thesaadmemon, 2022-10-17:
Do we have support for the ZCU102 board? @jwfaye and @enjoy-digital

jwfaye, 2023-01-12:
@thesaadmemon Hello, I manage to make work. the associated codes of the platforms are available in platform.py and target.py