enjoy-digital/litex #1244: SDcard boot fails at 50Mhz+ system clock on Arty A7

developandplay, 2022-03-18:
Tested on Rocket + Veriscv on Arty A7 using the sdcard PMOD from digilent.

In SPI sdcard mode the boot files don't get detected.
If I lower the system clock to 20Mhz then SPI mode works for booting.
If I enable ethernet at 20Mhz then the files get detected but when copying the system freezes.

Regular sdcard mode doesn't work at all in any of the configurations I tried.

Honestly I have no idea what could be causing the behaviour I described.
My original aim was to use Rocket at 50Mhz with regular sdcard and ethernet.

developandplay, 2022-03-19:
I tried Vexriscv SMP @100Mhz and didn't get success with the SDcard.
`digilent_arty.py --build --load --cpu-type vexriscv_smp --sys-clk-freq 100e6 --with-sdcard --variant=a7-100`

For `--with-sdcard` the binary doesn't get detected.
For `--with-spi-sdcard` the system freezes while copying the binary.

developandplay, 2022-03-19:
Went all the way back to commit c3b94a347e1e4a14c11b82473a0951de81939477
Back then both sdcard modes were working fine.
Will see if I can narrow it down to a certain commit.

enjoy-digital, 2022-03-20:
Could you try reverting https://github.com/enjoy-digital/litesdcard/commit/809ade33137edd6db59912844b4a7ccfbe2e4e0c? I did a test on arty before integrating it, but this  is the most probable change that could explain a difference in the behavior.

developandplay, 2022-03-20:
I don't think that's the issue as I only reverted litex to https://github.com/enjoy-digital/litex/commit/c3b94a347e1e4a14c11b82473a0951de81939477 and the litex-boards to a commit from that day.

I left litesdcard untouched.

developandplay, 2022-03-20:
Nevermind the issue was on my side.
I previously had disabled the `-flto` flag from the bios to get the litex `demo.bin` to work.
(I now tried a newer toolchain version which also gives me the error `lto-wrapper failed` on compilation of `demo.bin` if `-flto` is enabled.)

Then I didn't add it back again later on. If I add back `-flto` sdcard boot works as expected for bbl (not for the demo though).

closed 2022-03-20 by developandplay

developandplay, 2022-03-20:
Oh and I just tried compiling BIOS and demo bin with the most recent [riscv toolchain build](https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2022.03.09) which also works now without removing the `-flto` flag.

In addition there some minor fixes required: #1245