enjoy-digital/litex #1253: litex_term image load

controlpaths, 2022-03-25:
Hello, 

I am trying to load the linux image of linux on litex using litex_term function, but the command is stuck. Debugging the python code I discover than the line where it is stucked is here (litex_term.py)

    term.join(True)

in the join function

    def join(self, writer_only=False):
        self.writer_thread.join()
        if not writer_only:
            self.reader_thread.join()

The command I am executing is the next

litex_term --images=images/boot.json /dev/ttyUSB1

Few month ago I use this command without issues using a SERV core loading a kernel instead of an image. The litex local repository is in the head of the remote...

thanks!

enjoy-digital, 2022-03-26:
Hello,

can you also provide the build command for the gateware? Thanks.

controlpaths, 2022-03-26:
Hello, sure! 

./make.py --board arty --cpu-count 2 --build
./make.py --board arty --cpu-count 2 --load

I have tried without the argument --cpu-count ant it has go ahead until the image load. Then an error rise up because it cannot found the image file.

The entire boot process is the next:

~/linux-on-litex-vexriscv$ litex_term --images=images/boot.json /dev/ttyUSB1

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS CRC passed (80be64f2)

 Migen git sha1: ac70301
 LiteX git sha1: a0853d15

--=============== SoC ==================--
CPU:		VexRiscv SMP-LINUX @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		64KiB
SRAM:		6KiB
FLASH:		16384KiB
SDRAM:		262144KiB 16-bit @ 800MT/s (CL-7 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |11111111111000000000000000000000| delays: 05+-05
  m0, b03: |00000000000011111111111111000000| delays: 19+-07
  m0, b04: |00000000000000000000000000001111| delays: 30+-02
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b03 delays: 19+-07
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |11111111111000000000000000000000| delays: 05+-05
  m1, b03: |00000000000011111111111111000000| delays: 19+-07
  m1, b04: |00000000000000000000000000001111| delays: 30+-02
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b03 delays: 19+-07
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 164.2MiB/s
   Read speed: 78.0MiB/s

Initializing S25FL128L SPI Flash @0x01000000...
Enabling Quad mode...
SPI Flash clk configured to 100 MHz
Memspeed at 0x1000000 (Sequential, 4.0KiB)...
   Read speed: 22.2MiB/s
Memspeed at 0x1000000 (Random, 4.0KiB)...
   Read speed: 5.3MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pablo/litex/litex/litex/tools/litex_term.py", line 544, in reader
    self.answer_magic()
  File "/home/pablo/litex/litex/litex/tools/litex_term.py", line 530, in answer_magic
    self.upload(filename, int(base, 16))
  File "/home/pablo/litex/litex/litex/tools/litex_term.py", line 431, in upload
    f = open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: 'images/Image'

The image that litex_term is using is not the image I have configured... I can't found in the code where is the image changed. 

Thanks for your help!

enjoy-digital, 2022-03-26:
It seems that the images are not copied correctly in `images` directory.

controlpaths, 2022-03-26:
The process to copy the images in that directory is made by the make.py script, is that correct?

in my case, the content of the images directory is the next

~/linux-on-litex-vexriscv/images$ tree
.
├── boot.json
└── rv32.dtb

I think that some files are missing. Where exactly are created the image files?

thanks for the support!

controlpaths, 2022-03-26:
Hello, 
Finally I been able to boot linux using the images from here 
https://github.com/litex-hub/linux-on-litex-vexriscv/issues/164
I am going to test again using the argument --cpu-core=2. 

Thanks!

controlpaths, 2022-03-26:
hello, 

When I configure 2 cpu cores, the image load is stucked, It does not happens without the argument. I have tried to rebuild all from the tag 2021.12 of Litex, with the same result. 

Thanks!

enjoy-digital, 2022-03-27:
Hello,

It has been tested recently and should work. I'll do a test tomorrow.

controlpaths, 2022-03-27:
I am using the python version 3.8.10. I will check again all the dependencies and the entire process to be sure that it is not my mistake.

Thanks!

enjoy-digital, 2022-03-28:
@controlpaths: Sorry, the issue was in fact related to a regression introduced by https://github.com/enjoy-digital/litex/commit/967ea12364c3ff307503b5aea17dad701dd5f69c and that has just been fixed with https://github.com/enjoy-digital/litex/commit/e8ed29776345a119c9a2e094915ba04793c9590b. I just tested it locally on an Arty.

enjoy-digital, 2022-03-29:
Closing since reproduced and fixed, please re-open @controlpaths if you still have troubles.

closed 2022-03-29 by enjoy-digital