enjoy-digital/litex #1260: microwatt BIOS hangs in ISR

madscientist159, 2022-04-01:
With the current LiteX versions and the Microwatt CPU (`--cpu-type=microwatt --cpu-variant=standard+ghdl+irq`) the BIOS hangs in the ISR with what appears to be a wild jump to an invalid address outside of the memory space.  This in turn results in no console output and is repeatable in simulation.

I've attached a simulation trace database and the BIOS files generated from a Debian Bullseye ppc64le host system for analysis.  The last instructions executed appear to be as follows:

```
        /* Load IRQ handler address from SRAM */
        LOAD_IMM64(%r3, __isr_address)
    102c:       00 00 60 3c     lis     r3,0
    1030:       00 00 63 60     ori     r3,r3,0
    1034:       c6 07 63 78     rldicr  r3,r3,32,31
    1038:       00 00 63 64     oris    r3,r3,0
    103c:       00 00 63 60     ori     r3,r3,0
        ld      %r3,   0(%r3)
    1040:       00 00 63 e8     ld      r3,0(r3)

        mtctr   %r3,
    1044:       a6 03 69 7c     mtctr   r3
        mr      %r3, %r10
    1048:       78 53 43 7d     mr      r3,r10
        bctrl
    104c:       21 04 80 4e     bctrl
        nop
    1050:       00 00 00 60     nop
        ld      %r1, 0(%r1)
    1054:       00 00 21 e8     ld      r1,0(r1)

        ld      %r0,   1*8(%r1)
    1058:       08 00 01 e8     ld      r0,8(r1)
//      ld      %r1,   1*8(%r1) // do this at rfid
        ld      %r2,   2*8(%r1)

<jump to invalid address 0x4800002408000048>
```
[simulation_output_and_bios_files.zip](https://github.com/enjoy-digital/litex/files/8399156/simulation_output_and_bios_files.zip)

madscientist159, 2022-04-01:
@shenki Any chance you might be able to take a look at this?

madscientist159, 2022-04-01:
From what I can tell, objdump might be mistranslating `bctrl`, microwatt is processing that instruction as a `bcctr` (not sure which is correct right now, so I'm assuming microwatt is doing the right thing).

If true, the invalid jump is caused by `ld      r3,0(r3)` loading the data at 0x0 into r3, which is obviously incorrect.  Starting to suspect `__isr_address` might be wrong at this point.

madscientist159, 2022-04-01:
Confirmed, LOAD_IMM64(%r3, __isr_address) is loading 0x0 (ROM base) instead of the correct location 0x100000 (SRAM base).

When run without IRQ support the console does come up, but there are residual issues that may be caused by related linking errors.  Will continue to debug here.

madscientist159, 2022-04-01:
OK, it appears this has been an issue for quite a while and no one was testing, including on our side, as we were using a different boot firmware that takes advantage of the `SPR_KAISB` feature.

I'll be spending some time to apply https://gitlab.raptorengineering.com/kestrel-collaboration/kestrel-firmware/bootrom/-/merge_requests/10 to the LiteX tree and creating a new pull request once I know it's working...

enjoy-digital, 2022-04-05:
Thanks @madscientist159, it's planned to add Microwatt testing to the CI (at it is currently done for most supported RISC-V CPUs), but this hasn't been done yet but we could put more efforts into this.