enjoy-digital/litex #1263: Regression - BlackParrot+Arty demo.bin stuck on lift-off

developandplay, 2022-04-01:
I just tried if I can reproduce https://github.com/enjoy-digital/litex/pull/1259 on BlackParrot.
Unfortunately using BlackParrot and the Arty board the `demo.bin` gets stuck on lift-off even if `-flto is removed`.
Here are the commands that I used:
```
digilent_arty.py --build --load --sys-clk-freq 20e6 --cpu-type blackparrot --cpu-variant standard --variant=a7-100
litex_bare_metal_demo --build-path=build/digilent_arty
lxterm /dev/ttyUSB1 --kernel demo.bin --kernel-adr 0x80000000
sdram_mr_write 0 2624
sdram_mr_write 1 2054
sdram_mr_write 2 512
sdram_cal
serialboot
```

Interestingly in simulation the demo application works as expected (at least without `-flto`):
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr  2 2022 00:09:09
 BIOS CRC passed (3ca1a04e)

 Migen git sha1: ac70301
 LiteX git sha1: ab9ddc99

--=============== SoC ==================--
CPU:		BlackParrotRV64[imafd] @ 1MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		65536KiB 32-bit @ 1MT/s (CL-2 CWL-2)

--========== Initialization ============--
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x80000000

--============= Liftoff! ===============--

LiteX minimal demo app built Apr  2 2022 00:09:05

Available commands:
help               - Show this command
reboot             - Reboot CPU
donut              - Spinning Donut demo
helloc             - Hello C
litex-demo-app> 
```

enjoy-digital, 2022-04-05:
Hi @developandplay,

20MHz is quite low, is the DDR3 calibrating correctly at this freq and memtest passing? If so, have you tested the same design also at 20MHz with Rocket?

I would recommend first trying to execute  the `demo.bin` from an integrated RAM by adding` --integrated-main-ram-size=0x10000` (Adjust size to your needs). This will remove the DDR3 from the equation. If working like this, the issue is the` sys_clk_freq` that is too low, are you able to increase it of is 20MHz the maximum frequency for BlackParrot on Arty?

developandplay, 2022-04-06:
So the memory is calibrated and the test is passing at 20Mhz. I have been
using this setup previously that is why I marked it as a regression.
I'll try rocket at 20Mhz and the integrated ram as well.
Besides dram are there frequency dependent components that might cause this?