enjoy-digital/litex #1264: Zephyr on Litex/Vexriscv not booting

lachlansmith, 2022-04-02:
**Describe the bug**
A clear and concise description of what the bug is.

I'm looking for help with Litex Vexriscv on an Arty A7 35T. I'm unable to boot zephyr no matter what I try and I'm wondering if I'm missing something fundamental. I'm struggling with what documentation is out there. I'll more than happily write documentation if someone could help me here today.

After loading the BIOS onto the board any build with litex_vexriscv to create the zephyr.bin files leaves the terminal blank after liftoff. However, I able to load the bare metal demo.bin file successfully so I'm stumped as what I'm doing wrong.

To Reproduce
Steps to reproduce the behavior:
Here's everything I did from the beginning.
```
mkdir litex
cd litex
wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
chmod +x litex_setup.py
./litex_setup.py --init --install --user
pip3 install meson ninja
./litex_setup.py --gcc=riscv
mv riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux-ubuntu14 riscv64-unknown-elf
export PATH="/home/lachy/litex/riscv64-unknown-elf/bin:$PATH"
sudo apt install libevent-dev libjson-c-dev verilator
cd litex-boards/litex_boards/targets/
./digilent_arty.py --toolchain vivado --cpu-type vexriscv --sys-clk-freq 80e6 --build
./digilent_arty.py --load
cd ~/zephyr/samples/subsys/shell/shell_module
west build -p auto -b litex_vexriscv .
litex_term /dev/ttyUSB1 --kernel build/zephyr/zephyr.bin --serial-boot

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr  2 2022 11:32:31
 BIOS CRC passed (8bdcbcdb)

 Migen git sha1: ac70301
 LiteX git sha1: c4004791

--=============== SoC ==================--
CPU:            VexRiscv @ 80MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          262144KiB 16-bit @ 640MT/s (CL-7 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |11110000000000000000000000000000| delays: 02+-02
  m0, b02: |00000011111111111111111100000000| delays: 15+-09
  m0, b03: |00000000000000000000000000011111| delays: 29+-03
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b02 delays: 15+-09
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |11110000000000000000000000000000| delays: 02+-02
  m1, b02: |00000011111111111111111100000000| delays: 15+-09
  m1, b03: |00000000000000000000000000011111| delays: 29+-02
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: 15+-09
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 29.2MiB/s
   Read speed: 39.0MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading build/zephyr/zephyr.bin to 0x40000000 (187436 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (9.9KB/s).
[LITEX-TERM] Booting the device.
[LITEX-TERM] Done.
Executing booted program at 0x40000000

--============= Liftoff! ===============--
```
**Expected behavior**
A clear and concise description of what you expected to happen.
```
--============= Liftoff! ===============--

uart:~$
```

lachlansmith, 2022-04-02:
Needed to generate dts, https://github.com/zephyrproject-rtos/zephyr/issues/44478. As a note, couldn't find documentation on this. Could someone link that here perhaps to assist others that are starting out like myself.

enjoy-digital, 2022-04-04:
@lachlansmith: Thanks, would you mind checking if https://github.com/enjoy-digital/litex/wiki/Run-Zephyr-On-Your-SoC seems up to date for you? If not, we could update, add more details. Have you also been able to find this Wiki page while looking at the issue? If not, we could add more links to it.

lachlansmith, 2022-04-05:
Hi @enjoy-digital thanks for getting back to me. 

The documentation could use an update I believe. [Run-Zephyr-On-Your-SoC](https://github.com/enjoy-digital/litex/wiki/Run-Zephyr-On-Your-SoC) assumes the user is familiar with all of build, load and flash steps. It took me some time to understand the role of the BIOS, and that --flash wasn't what I was looking for when trying to flash Zephyr. 

The litex_term step I think is quite important final step for beginners, and is overlooked in a lot of the documentation out there. It now seems obvious to me, but it very much wasn't at the start.
```
litex_term /dev/ttyUSB1 --kernel build/zephyr/zephyr.bin --serial-boot
``` 
Other than a brief explanation/link to how to use the **LiteXSoCArgumentParser** and addition of **litex_term** step the documentation is exactly what is needed. I think your idea to include more back links to the Wiki would be great, it has horrible SEO.

mckaymatthew, 2022-04-09:
I do believe this is a duplicate of https://github.com/enjoy-digital/litex/issues/1062