enjoy-digital/litex #1266: Dlc9 with cle-215

vladoshub, 2022-04-03:
hello. Can anyone tell me if i can use DLC9 to flash my acorn board?

enjoy-digital, 2022-04-05:
Hi @vladoshub,

IIRC I've been able to reprogram the Acorn through a JTAG HS2 and Vivado Hardware Manager, so I assume it should also be possible with a DLC9 cable. Can you give it a try? The SPI Flash on the Acorn is a S25FL128.

vladoshub, 2022-04-05:
> Hi @vladoshub,
> 
> IIRC I've been able to reprogram the Acorn through a JTAG HS2 and Vivado Hardware Manager, so I assume it should also be possible with a DLC9 cable. Can you give it a try? The SPI Flash on the Acorn is a S25FL128.

ok, as I understand it, you just need to build the project, and take the files and use vivado for flash. Can you tell me what files I need for this, except for the .bin file?

enjoy-digital, 2022-04-05:
Yes, you can just build the Acorn target with `--build`, then you'll have the files in `build/sqrl_acorn/gateware`. Open Vivado Hardware Manager, first try to load the `.bit `in SRAM and then you could try to Flash the `.bin` in the SPI Flash.

vladoshub, 2022-04-05:
> Yes, you can just build the Acorn target with `--build`, then you'll have the files in `build/sqrl_acorn/gateware`. Open Vivado Hardware Manager, first try to load the `.bit` in SRAM and then you could try to Flash the `.bin` in the SPI Flash.

can I flash with pcie in the future? I just saw that there is also a file that indicates the range of flashing in memory, or will a .bin file for spi be enough and in the future I can use the command

`./litepcie_util flash_write acorn_pcie_operational.bin 0x400000`

enjoy-digital, 2022-04-05:
To be able to flash over PCIe, you first need the LiteX bitstream to be loaded, compiled with LitePCIe (--with-pcie), the board installed in a PCIe port and LitePCIe kernel driver loaded.

vladoshub, 2022-04-08:
> To be able to flash over PCIe, you first need the LiteX bitstream to be loaded, compiled with LitePCIe (--with-pcie), the board installed in a PCIe port and LitePCIe kernel driver loaded.

Thanks for the help, surprisingly everything went smoothly, I flashed SRAM and then SPI, except that I have S25FL256. Unfortunately, I did not find a JTAG cable for sale and I had to solder the wires directly to the connector. Can you tell me how fallback and operational bitstream work, let's say I want to change the design and play around with leds first, and what should I not do so that I can always use pcie? as I understand it, I won’t be able to flash the board if I change the pcie bridge, and in case of an error
 boot the backup part that is flashed into spi for the very first time

enjoy-digital, 2022-04-08:
Hi @vladoshub,

you should be able to find the information here: https://github.com/enjoy-digital/litex/wiki/Use-LiteX-on-the-Acorn-CLE-215#acorn-giveaway. You can also read: https://docs.xilinx.com/v/u/en-US/xapp1247-multiboot-spi.