enjoy-digital/litex #1272: Support custom toolchain

xhebox, 2022-04-12:
I am on a musl-libc based machine, thus I prefer to compile a custom toolchain by myself. A summary of problems:

1. `xxx-pc-linux-musl` triple in toolchain detection
2. After a version of binutils(haven't looked into it), `zicsr` are separated from `i`, which implies `-march=rv32i_zicsr` to correct compile softwares. BTW, I am binutils-2.28. Maybe some post-processing to cflags with version detection.
3. picolibc has many `__need_xxx_t` macro, while musl only support to detect `__NEED_xxx_t`. Also musl will define `DEFINED_size_t` instead of `_SIZE_T`. Maybe this should be reported to picolibc instead..
4. Compile picolibc with musl will results in `undefined reference 'write' in __chk_fail`, unknown cause.
5. Custom toolchain from archlinux or mine, can not show the bios menu correctly, unknown cause.

I suspect 4 and 5 may have the same root cause, still investigating it.

mithro, 2022-04-12:
@xhebox - You need to be careful about the difference between the toolchain being used for the **target** and the toolchain you use for the **host**. I don't believe anyone is currently running musl on a LiteX based system (but I could be wrong).

xhebox, 2022-06-09:
@enjoy-digital Could you reopen the issue? Since https://github.com/enjoy-digital/litex/commit/eab58ff0e9bc707c2809261b44d907e809057796 has reverted the working patch.

xhebox, 2022-06-14:
ping @enjoy-digital