enjoy-digital/litex #1285: video_framebuffer: selecting format rgb565 results in a crash duing verilog generation

swetland, 2022-04-27:
To try to save DP16KB memories (the default video_framebuffer configuration uses 32), I attempted to choose a 16bpp format ("rgb565") in hopes of cutting the fifo size in half.

I modified radiona_ulx3s.py, changing:
```self.add_video_framebuffer(phy=self.videophy, timings="640x480@75Hz", clock_domain="hdmi")```
to
```self.add_video_framebuffer(phy=self.videophy, timings="640x480@75Hz", clock_domain="hdmi", format="rgb565")```

```$ ./radiona_ulx3s.py --with-video-framebuffer
...
Traceback (most recent call last):
  File "./radiona_ulx3s.py", line 198, in <module>
    main()
  File "./radiona_ulx3s.py", line 191, in main
    builder.build(**builder_kargs, run=args.build)
  File "/work/litex-install/litex/litex/soc/integration/builder.py", line 350, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/work/litex-install/litex/litex/soc/integration/soc.py", line 1198, in build
    return self.platform.build(self, *args, **kwargs)
  File "/work/litex-install/litex/litex/build/lattice/platform.py", line 43, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/work/litex-install/litex/litex/build/lattice/trellis.py", line 215, in build
    v_output = platform.get_verilog(fragment, name=build_name, **kwargs)
  File "/work/litex-install/litex/litex/build/lattice/platform.py", line 36, in get_verilog
    return GenericPlatform.get_verilog(self, *args,
  File "/work/litex-install/litex/litex/build/generic_platform.py", line 423, in get_verilog
    return verilog.convert(fragment, platform=self, **kwargs)
  File "/work/litex-install/litex/litex/gen/fhdl/verilog.py", line 602, in convert
    verilog += _print_combinatorial_logic_synth(f, ns)
  File "/work/litex-install/litex/litex/gen/fhdl/verilog.py", line 471, in _print_combinatorial_logic_synth
    r += _print_node(ns, _AT_NONBLOCKING, 1, g[1])
  File "/work/litex-install/litex/litex/gen/fhdl/verilog.py", line 287, in _print_node
    return "".join(_print_node(ns, at, level, n, target_filter) for n in node)
  File "/work/litex-install/litex/litex/gen/fhdl/verilog.py", line 287, in <genexpr>
    return "".join(_print_node(ns, at, level, n, target_filter) for n in node)
  File "/work/litex-install/litex/litex/gen/fhdl/verilog.py", line 292, in _print_node
    r += _print_node(ns, at, level + 1, node.t, target_filter)
  File "/work/litex-install/litex/litex/gen/fhdl/verilog.py", line 287, in _print_node
    return "".join(_print_node(ns, at, level, n, target_filter) for n in node)
  File "/work/litex-install/litex/litex/gen/fhdl/verilog.py", line 287, in <genexpr>
    return "".join(_print_node(ns, at, level, n, target_filter) for n in node)
  File "/work/litex-install/litex/litex/gen/fhdl/verilog.py", line 283, in _print_node
    return "\t"*level + _print_expression(ns, node.l)[0] + assignment + _print_expression(ns, node.r)[0] + ";\n"
  File "/work/litex-install/litex/litex/gen/fhdl/verilog.py", line 249, in _print_expression
    return _print_slice(ns, node)
  File "/work/litex-install/litex/litex/gen/fhdl/verilog.py", line 212, in _print_slice
    assert (node.stop - node.start) >= 1
AssertionError
```

enjoy-digital, 2022-04-28:
Thanks @swetland, in fact you just need to update LiteX :) since this has been fixed a few days ago with https://github.com/enjoy-digital/litex/commit/625261c69301a5c3a310f7b36ea959fa0a0828bb.

closed 2022-04-28 by enjoy-digital

swetland, 2022-04-28:
Nice!  Confirmed working in tip of tree. Thanks!

qee-d, 2024-10-31:
self.add_video_framebuffer(phy=self.videophy, timings="800x600@60Hz", clock_domain="hdmi")，默认的颜色格式是什么？rgb888，还是rgba？