enjoy-digital/litex #1296: --csr-json csr.json  doesn't produce the output file

mzau, 2022-05-12:
I installed litex as instructed  especially installing the Risc-V toolchain with

`./litex_setup.py --gcc=riscv`

Also 
`litex_sim --cpu-type=vexriscv`
is starting  up a simulated litex Bios.

Though following the wiki at [Generate Renode Simulation](https://github.com/enjoy-digital/litex/wiki/Generate-Renode-simulation)

`python3 litex-boards/litex_boards/targets/digilent_arty.py --cpu-type vexriscv --with-ethernet --csr-json csr.json`

Runs through without flagging any error, **but the `csr.json`  file is not generated.**

Where do I find the csr.json which I need to generate a .repl for renode?

mzau, 2022-05-12:
adding  `--build --no-compile` seemes to do it.

enjoy-digital, 2022-05-12:
Hi @mzau,

thanks for reporting. This is indeed a side-effect of disabling the build by default (which also had other side-effect), that I also encountered a few hours ago. I'll try to improve this.

fdsgdshn, 2023-06-07:
Hi, 

Is there any progress on this issue? I'm also encountering the same error when generating the Litex.resc and litex.repl files for Renode. I'm getting the "config_cpu_count" error, which indicates that this attribute is not correctly defined in the csr.json file. It seems that the "--build-no-compile" command is not working correctly.