enjoy-digital/litex #1297: STLV-7325 Rocket sdcardboot issue

developandplay, 2022-05-12:
When trying to boot Rocketchip from the sdcard on the stlv-7325 board I get stuck on lift-off.
Using Vexriscv and the stlv-7325 board I can get sdcardboot to work at 100Mhz (with l2 disabled).
Using Rocketchip and the Arty A7 I can get sdcardboot to work at 50Mhz.

I tried both frequencies with the stlv-7325 and timing are met for both.
This is the command that I'm using:
```
../litex/litex-boards/litex_boards/targets/aliexpress_stlv7325.py --build --cpu-type rocket --cpu-variant linux2d --sys-clk-freq 100e6 --with-sdcard --with-ethernet --csr-csv=csr.csv --output-dir build/aliexpress_stlv7325-100
```

I'm a bit confused since I can't reproduce this on any other setup.
Any debugging advice would be greatly appreciated.

developandplay, 2022-05-12:
Seems like email response didn't work.

`--with-spi-sdcard` was not working either for Rocket at 50Mhz/100Mhz on the stlv-7325.
I forgot to add: Using Rocket + STLV-7325 + serialboot works as expected.
VexRiscv + STLV-7325 + sdcardboot is working at 50Mhz, 100Mhz, 150Mhz and 175Mhz (all with memory callibration and without l2).

Is there anything that could be specific to Rocket+STLV-7325+sdcard?
Maybe something that is (not) present on the Arty board or something that vexriscv is (not) using?
I will try the reverted commit next.
Thanks for the advice.

enjoy-digital, 2022-05-12:
I don't see obvious reasons it would work on Arty at 50MHz and not on the SLV7325. If you want to have a configuration as close a possible to the Arty, you could eventually try to reduce the DDR3 to only 16-bit with the `PHYPadsReducer`:
https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/xilinx_ac701.py#L71

Use `PHYPadsReducer(platform.request("ddram"), [0, 1])` for 16-bit.

developandplay, 2022-05-13:
I tried reverting https://github.com/enjoy-digital/litesdcard/commit/809ade33137edd6db59912844b4a7ccfbe2e4e0c which didn't seem to make any difference.

I'll try reducting DDR3 to 16-bit width.

I just hope it's not some stupid Vivado bug like the one for my Bachelor thesis that took a month to debug.

(It was about Vivado handling sign extension improperly but only in the toolchain for the Artix series and not in the toolchain for the Kintex series.)

developandplay, 2022-05-28:
Reducing to 16bit width didn't make any difference. I'll also check if ethernet boot works.