enjoy-digital/litex #1298: --cpu-count option not working for vexriscv_smp

hansfbaier, 2022-05-16:
```
python3 litex_boards/targets/hpcstore_xc7k420t.py --cpu-type vexriscv_smp --cpu-count=16 --build 
usage: hpcstore_xc7k420t.py [-h] [--build] [--load] [--sys-clk-freq SYS_CLK_FREQ] [--io-voltage IO_VOLTAGE]
                            [--with-pcie] [--driver] [--with-sata] [--output-dir OUTPUT_DIR]
                            [--gateware-dir GATEWARE_DIR] [--software-dir SOFTWARE_DIR] [--include-dir INCLUDE_DIR]
                            [--generated-dir GENERATED_DIR] [--no-compile] [--no-compile-software]
                            [--no-compile-gateware] [--csr-csv CSR_CSV] [--csr-json CSR_JSON] [--csr-svd CSR_SVD]
                            [--memory-x MEMORY_X] [--doc] [--bus-standard BUS_STANDARD]
                            [--bus-data-width BUS_DATA_WIDTH] [--bus-address-width BUS_ADDRESS_WIDTH]
                            [--bus-timeout BUS_TIMEOUT] [--bus-bursting] [--cpu-type CPU_TYPE]
                            [--cpu-variant CPU_VARIANT] [--cpu-reset-address CPU_RESET_ADDRESS] [--cpu-cfu CPU_CFU]
                            [--no-ctrl] [--integrated-rom-size INTEGRATED_ROM_SIZE]
                            [--integrated-rom-init INTEGRATED_ROM_INIT] [--integrated-sram-size INTEGRATED_SRAM_SIZE]
                            [--integrated-main-ram-size INTEGRATED_MAIN_RAM_SIZE] [--csr-data-width CSR_DATA_WIDTH]
                            [--csr-address-width CSR_ADDRESS_WIDTH] [--csr-paging CSR_PAGING]
                            [--csr-ordering CSR_ORDERING] [--ident IDENT] [--no-ident-version] [--no-uart]
                            [--uart-name UART_NAME] [--uart-baudrate UART_BAUDRATE] [--uart-fifo-depth UART_FIFO_DEPTH]
                            [--no-timer] [--timer-uptime] [--l2-size L2_SIZE]
hpcstore_xc7k420t.py: error: unrecognized arguments: --cpu-count=16
python3 litex_boards/targets/hpcstore_xc7k420t.py --cpu-type vexriscv_smp    0,26s user 0,02s system 99% cpu 0,281 total
```

enjoy-digital, 2022-05-16:
@hansfbaier: This is working here, are you sure you are using upstream LiteX?

cklarhorst, 2022-05-17:
Thanks hansfbaier! 
I had the same problem a few times, but it was sometimes working and on other days not.

The problem lies in the parsing code: https://github.com/enjoy-digital/litex/blob/71a5ef238087de5fb973f5fe7ad31b29d67bd07e/litex/soc/integration/soc.py#L1990

Argparser is happy to parse "--cpu-type vexriscv_smp" but LiteX expects it to be "--cpu-type=vexriscv_smp"!

enjoy-digital, 2022-05-17:
@cklarhorst: Time to fix this :) Thanks!

cklarhorst, 2022-05-17:
@enjoy-digital
How do you like the following idea (2 stage parsing):

Use parse_known_args to preparse the arguments for the cpu_type.
This would exit on help arguments, so we need add_help to False on our parser.
After that, we do the normal parsing with parse_args. 
We have to create a new ArgumentParser instance for this parse_args call in order to get the help arguments back.
So we create the new (second) instance with the parents option set to our old parser and with add_help to true.

Full POC Code:
```
class LiteXSoCArgumentParser(argparse.ArgumentParser):
    def __init__(self, prog=None, usage=None, description=None, epilog=None, parents=[], 
                 formatter_class=argparse.HelpFormatter, prefix_chars='-', 
                 fromfile_prefix_chars=None, argument_default=None, 
                 conflict_handler='error', add_help=True, allow_abbrev=True, 
                 exit_on_error=True):
        self.add_help = add_help
        argparse.ArgumentParser.__init__(self, prog=prog, usage=usage, description=description, epilog=epilog, parents=parents,
                 formatter_class=formatter_class, prefix_chars=prefix_chars, fromfile_prefix_chars=fromfile_prefix_chars, 
                 argument_default=argument_default, conflict_handler=conflict_handler, add_help=False, allow_abbrev=allow_abbrev,
                 exit_on_error=exit_on_error)

    def parse_args(self):
        def get_selected_cpu_name():
            args, leftovers = argparse.ArgumentParser.parse_known_args(self)
            return cpu.CPUS.get(args.cpu_type, None)

        # Intercept selected CPU to fill arguments.
        cpu_cls = get_selected_cpu_name()
        if cpu_cls is not None and hasattr(cpu_cls, "args_fill"):
            cpu_cls.args_fill(self)

        # Get Command-line arguments.
        second_parser = argparse.ArgumentParser(parents=[self], add_help=self.add_help)
        args = second_parser.parse_args()

        # Re-inject CPU read arguments.
        if cpu_cls is not None and hasattr(cpu_cls, "args_read"):
            cpu_cls.args_read(args)

        return args
```
I would be happy to create a PR if that is in your favor.

enjoy-digital, 2022-05-17:
Thanks @cklarhorst, I just pushed a simple change with https://github.com/enjoy-digital/litex/commit/4810cc15dafccd07a0ae0ba1f01ed09b0fe6d07c. We could switch to a real 2 stage parser in the future if we have a need to do this for other parameters.

closed 2022-05-17 by enjoy-digital