enjoy-digital/litex #1313: CVA6 and AXILiteDownConverter

suppamax, 2022-05-25:
I think the problem mentioned in the PR #1294 and regarding the need to press a key to let the simulation run relates to this

1. the core executes `uart_sync()`, which means polling the address `(CSR_BASE + 0x1804L)`
2. the CPU interface is 8bytes aligned while the CSR is 4 bytes aligned
3. the read request is converted into 2 reads, using offset 0x1804 and 0x1808 
4. the requested data in the lower half of the databus
5. the requested address is 4 bytes aligned, the CPU has an 8bytes databus, so it expects the data on the higher half of the bus

Masking out the last 3 bits of A*ADDR in `cva6_wrapper.sv` works
```
    assign AWADDR_o   = {master[cva6_wrapper_pkg::External].aw_addr[AxiAddrWidth-1:3], 3'b000};
...
    assign ARADDR_o   = {master[cva6_wrapper_pkg::External].ar_addr[AxiAddrWidth-1:3], 3'b000};
```

Is this acceptable?

enjoy-digital, 2022-05-27:
Thanks for looking at this @suppamax, I'll have a closer look. This could be better to fix the AXILiteDownConverter's behavior if incorrect.

sergachev, 2022-07-24:
AXILiteDownConverter is not used here, it's axi64 -> axi32 -> axilite32 -> wb32 with the first conversion being the problem.

One possible solution is here: https://github.com/enjoy-digital/litex/pull/1387

enjoy-digital, 2022-12-09:
The AXIDownConverter has been improved/fixed and now handle things correctly and allow CVA9 simulation.