enjoy-digital/litex #1314: Couldn't Install LiteX When Using Arch Linux

minexo79, 2022-05-26:
OS: Arch Linux
Python Version: 3.10.4

---
Error Log: 
```
[   0.405] Installing migen Git repository...
Defaulting to user installation because normal site-packages is not writeable
Obtaining file:///home/minexo79/CodeWorkspace/litex/migen
Requirement already satisfied: colorama in /usr/lib/python3.10/site-packages (from migen==0.9.2) (0.4.4)
Installing collected packages: migen
  Running setup.py develop for migen
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/minexo79/CodeWorkspace/litex/migen/setup.py'"'"'; __file__='"'"'/home/minexo79/CodeWorkspace/litex/migen/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps --user --prefix=
         cwd: /home/minexo79/CodeWorkspace/litex/migen/
    Complete output (28 lines):
    running develop
    /usr/lib/python3.10/site-packages/setuptools/command/easy_install.py:157: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
      warnings.warn(
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/minexo79/CodeWorkspace/litex/migen/setup.py", line 14, in <module>
        setup(
      File "/usr/lib/python3.10/site-packages/setuptools/__init__.py", line 155, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 148, in setup
        return run_commands(dist)
      File "/usr/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 163, in run_commands
        dist.run_commands()
      File "/usr/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 967, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 985, in run_command
        cmd_obj.ensure_finalized()
      File "/usr/lib/python3.10/site-packages/setuptools/_distutils/cmd.py", line 107, in ensure_finalized
        self.finalize_options()
      File "/usr/lib/python3.10/site-packages/setuptools/command/develop.py", line 52, in finalize_options
        easy_install.finalize_options(self)
      File "/usr/lib/python3.10/site-packages/setuptools/command/easy_install.py", line 276, in finalize_options
        self._fix_install_dir_for_user_site()
      File "/usr/lib/python3.10/site-packages/setuptools/command/easy_install.py", line 382, in _fix_install_dir_for_user_site
        self.create_home_path()
      File "/usr/lib/python3.10/site-packages/setuptools/command/easy_install.py", line 1338, in create_home_path
        if path.startswith(home) and not os.path.isdir(path):
    AttributeError: 'int' object has no attribute 'startswith'
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/minexo79/CodeWorkspace/litex/migen/setup.py'"'"'; __file__='"'"'/home/minexo79/CodeWorkspace/litex/migen/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' develop --no-deps --user --prefix= Check the logs for full command output.
Traceback (most recent call last):
  File "/home/minexo79/CodeWorkspace/litex/./litex_setup.py", line 442, in <module>
    main()
  File "/home/minexo79/CodeWorkspace/litex/./litex_setup.py", line 424, in main
    litex_setup_install_repos(config=args.config, user_mode=args.user)
  File "/home/minexo79/CodeWorkspace/litex/./litex_setup.py", line 274, in litex_setup_install_repos
    subprocess.check_call("\"{python3}\" -m pip install --editable . {options}".format(
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '"/usr/bin/python3" -m pip install --editable . ' returned non-zero exit status 1.
```

minexo79, 2022-05-27:
It's Solved! Just change setuptool to 60.7.0 could be avoid this error.

closed 2022-05-27 by minexo79

shaw-wu, 2024-10-27:
I have a similar problem which indicates "could not find a version that satisfies the requirement colorama (from migen) (from versions: none) ",and seems likely "subprocess.CalledProcessError" error type.

shaw-wu, 2024-10-27:
I have a similar problem which indicates "could not find a version that satisfies the requirement colorama (from migen) (from versions: none) ",and seems likely "subprocess.CalledProcessError" error type.

error log:

`Defaulting to user installation because normal site-packages is not writeable                                                      
Looking in indexes: http://pypi.douban.com/simple                                                                                  
Requirement already satisfied: colorama in /usr/lib/python3/dist-packages (0.4.4)                                                  
shaw@shaw:~/linux-on-litex-vexriscv$ ./litex_setup.py --init --install --user                                                      
          __   _ __      _  __                                                                                                     
         / /  (_) /____ | |/_/                                                                                                     
        / /__/ / __/ -_)>  <                                                                                                       
       /____/_/\__/\__/_/|_|                                                                                                       
     Build your hardware, easily!                                                                                                  
          LiteX Setup utility.                                                                                                     
                                                                                                                                   
[   0.001] LiteX Setup auto-update...                                                                                              
[   0.001] Initializing Git repositories...                                                                                        
[   0.001] --------------------------------                                                                                        
[   0.001] migen Git Repo already present.                                                                                         
[   0.001] pythondata-software-picolibc Git Repo already present.                                                                  
[   0.001] pythondata-software-compiler_rt Git Repo already present.                                                               
[   0.001] litex Git Repo already present.                                                                                         
[   0.001] liteiclink Git Repo already present.                                                                                    
[   0.001] liteeth Git Repo already present.                                                                                       
[   0.001] litedram Git Repo already present.                                                                                      
[   0.001] litepcie Git Repo already present.                                                                                      
[   0.001] litesata Git Repo already present.                                                                                      
[   0.001] litesdcard Git Repo already present.                                                                                    
[   0.001] litescope Git Repo already present.                                                                                     
[   0.001] litejesd204b Git Repo already present.                                                                                  
[   0.001] litespi Git Repo already present.                                                                                       
[   0.001] valentyusb Git Repo already present.                                                                                    
[   0.001] litex-boards Git Repo already present.                                                                                  
[   0.001] pythondata-misc-tapcfg Git Repo already present.                                                                        
[   0.001] pythondata-misc-usb_ohci Git Repo already present.                                                                      
[   0.001] pythondata-cpu-lm32 Git Repo already present.                                                                           
[   0.001] pythondata-cpu-mor1kx Git Repo already present.                                                                         
[   0.001] pythondata-cpu-naxriscv Git Repo already present.                                                                       
[   0.001] pythondata-cpu-serv Git Repo already present.                                                                           
[   0.001] pythondata-cpu-vexiiriscv Git Repo already present.                                                                     
[   0.001] pythondata-cpu-vexriscv Git Repo already present.                                                                       
[   0.001] pythondata-cpu-vexriscv-smp Git Repo already present.                                                                   
[   0.001] Installing Git repositories...                                                                                          
[   0.001] ------------------------------                                                                                          
[... cut ...]
    subprocess.check_call("\"{python3}\" -m pip install --editable . {options}".format(                                            
  File "/usr/local/lib/python3.10/subprocess.py", line 369, in check_call                                                          
    raise CalledProcessError(retcode, cmd)                                                                                         
subprocess.CalledProcessError: Command '"/usr/local/bin/python3" -m pip install --editable . --user' returned non-zero exit status 
1.
`

It's easy to solve by use "sudo" to give root permissions!

shaw-wu, 2024-10-27:
I have a similar problem which indicates "could not find a version that satisfies the requirement colorama (from migen) (from versions: none) ",and seems likely "subprocess.CalledProcessError" error type.

error log:

```
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: http://pypi.douban.com/simple
Requirement already satisfied: colorama in /usr/lib/python3/dist-packages (0.4.4)
shaw@shaw:~/linux-on-litex-vexriscv$ ./litex_setup.py --init --install --user
__ _ __ _ __
/ / () /___ | |//
/ /__/ / __/ -)> <
///_/_//|_|
Build your hardware, easily!
LiteX Setup utility.

[ 0.001] LiteX Setup auto-update...
[ 0.001] Initializing Git repositories...
[ 0.001] --------------------------------
[ 0.001] migen Git Repo already present.
[ 0.001] pythondata-software-picolibc Git Repo already present.
[ 0.001] pythondata-software-compiler_rt Git Repo already present.
[ 0.001] litex Git Repo already present.
[ 0.001] liteiclink Git Repo already present.
[ 0.001] liteeth Git Repo already present.
[ 0.001] litedram Git Repo already present.
[ 0.001] litepcie Git Repo already present.
[ 0.001] litesata Git Repo already present.
[ 0.001] litesdcard Git Repo already present.
[ 0.001] litescope Git Repo already present.
[ 0.001] litejesd204b Git Repo already present.
[ 0.001] litespi Git Repo already present.
[ 0.001] valentyusb Git Repo already present.
[ 0.001] litex-boards Git Repo already present.
[ 0.001] pythondata-misc-tapcfg Git Repo already present.
[ 0.001] pythondata-misc-usb_ohci Git Repo already present.
[ 0.001] pythondata-cpu-lm32 Git Repo already present.
[ 0.001] pythondata-cpu-mor1kx Git Repo already present.
[ 0.001] pythondata-cpu-naxriscv Git Repo already present.
[ 0.001] pythondata-cpu-serv Git Repo already present.
[ 0.001] pythondata-cpu-vexiiriscv Git Repo already present.
[ 0.001] pythondata-cpu-vexriscv Git Repo already present.
[ 0.001] pythondata-cpu-vexriscv-smp Git Repo already present.
[ 0.001] Installing Git repositories...
[ 0.001] ------------------------------
[ 0.001] Installing migen Git repository...
Looking in indexes: http://pypi.douban.com/simple
Obtaining file:///home/shaw/migen
Preparing metadata (setup.py) ... done
ERROR: Could not find a version that satisfies the requirement colorama (from migen) (from versions: none)
ERROR: No matching distribution found for colorama
Traceback (most recent call last):
File "/home/shaw/linux-on-litex-vexriscv/./litex_setup.py", line 492, in
main()
File "/home/shaw/linux-on-litex-vexriscv/./litex_setup.py", line 472, in main
litex_setup_install_repos(config=args.config, user_mode=args.user)
File "/home/shaw/linux-on-litex-vexriscv/./litex_setup.py", line 286, in litex_setup_install_repos
subprocess.check_call(""{python3}" -m pip install --editable . {options}".format(
File "/usr/local/lib/python3.10/subprocess.py", line 369, in check_call
raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '"/usr/local/bin/python3" -m pip install --editable . --user' returned non-zero exit status 1.
```

It's easy to solve by use "sudo" to give root permissions!