enjoy-digital/litex #1320: Install GHDL and Yosys to CI to be able to simulate Microwatt/NeoRV32 with Verilator.

enjoy-digital, 2022-06-03:


mithro, 2022-06-03:
Containers with these tools are available from https://github.com/hdl/containers 

We will hopefully soon have GHDL available through the conda environment provided by https://github.com/hdl/conda-eda too.

mithro, 2022-06-03:
@umarcor, Antmicro and the BYU team has been working on making sure that the full F4PGA toolchain (with GHDL + Surelog + Yosys + VPR + nextpnr) is also easily available.

mithro, 2022-06-03:
FYI - @kgugala / @acomodi

umarcor, 2022-06-11:
See https://github.com/chipsalliance/f4pga/pull/574 and https://github.com/chipsalliance/f4pga/pull/575.

```yml
  Test-VHDL:
    runs-on: ubuntu-latest

    steps:

      - name: 🧰 Checkout
        uses: actions/checkout@v3

      - name: 🚧 GHDL synth
        run: make -C test/vhdl/counter synth

      - name: 🚧 F4PGA Action (arty_35 | vhdl/counter)
        uses: ./action
        with:
          image: xc7/a50t
          cmd: |
            cd test/vhdl/counter
            f4pga build --flow arty_35.json
      - name: '📤 Upload artifact: Arty 35 bitstream'
        uses: actions/upload-artifact@v3
        with:
          name: arty_35-Bitstream-VHDL-Counter
          path: test/vhdl/counter/top.bit
```

```makefile
synth:
	docker run --rm \
	-v /$(shell pwd)://wrk -w //wrk \
	gcr.io/hdl-containers/ghdl \
	ghdl synth --std=08 --out=verilog counter.vhd -e Arty_Counter > top.v
```

enjoy-digital, 2022-06-15:
Thanks @umarcor. I'm not sure this was possible in the past with Microwatt, but it seems to be the case now, so will simplify the conversion flow.

umarcor, 2022-06-15:
The flow through the ghdl-yosys-plugin was ready-to-use in early 2020 (I met Anton at ORCONF2019 and we collaborated to have the CI of microwatt use GHDL, Yosys and nextpnr), while the `ghdl synth` command was added in early 2021. Since many projects took Microwatt's makefile (and https://github.com/antonblanchard/ghdl-yosys-blink) as a reference, the other flow went partially unnoticed for the last year.

Nevertheless, note that the flows are not exactly equivalent. `ghdl synth` produces a non-optimised netlist, while yosys can run multiple stages. `ghdl synth` is mostly useful for linting and to work around tools with poor VHDL support.

closed 2022-11-21 by enjoy-digital