enjoy-digital/litex #1337: No output while running demo.bin on Litex SoC after Liftoff!

FATHY174, 2022-06-21:
I am trying to add new commands in the demo main code and then run it it gives me:

> /home/fathymd/litex2/riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux-centos6/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld:`_ demo.elf section `.data' will not fit in region `sram'
> /home/fathymd/litex2/riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux-centos6/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld:` region `sram' overflowed by 27408 bytes
> collect2:` error: ld returned 1 exit status
make: *** [Makefile:24: demo.elf] Error 1

so I go to the linker file and change the SRAM of the ".data" variable to main_ram, then it run 

but the problem is when I connect to the FPGA and boot the demo file it gives:

      __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun 21 2022 16:38:30
 BIOS CRC passed (ebe5b6f6)

 LiteX git sha1: a977adf5

--=============== SoC ==================--
CPU:		VexRiscv @ 125MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		524288KiB 32-bit @ 1000MT/s (CL-7 CWL-6)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 24
  Cmd/Clk scan (0-12)
  |101000111111| best: 2
  Setting Cmd/Clk delay to 2 taps.
  Data scan:
  m0: |11100000000000001111111111| delay: 16
  m1: |11110000000000000111111111| delay: 17
  m2: |11111100000000000011111111| delay: 18
  m3: |11111110000000000000111111| delay: 00
Write latency calibration:
m0:0 m1:0 m2:0 m3:6 
Write DQ-DQS training:
m0: |00000000000111111100000000000| delays: 16+-05
m1: |00000000000011111100000000000| delays: 16+-04
m2: |00000000000001111110000000000| delays: 17+-04
m3: |11000000000000000000000000000| delays: 02+-02
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00001111111110000000000000000000| delays: 08+-04
  m0, b04: |00000000000000000111111110000000| delays: 20+-03
  m0, b05: |00000000000000000000000000000011| delays: 00+-02
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b03 delays: 08+-04
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |10000000000000000000000000000000| delays: 02+-02
  m1, b03: |00001111111110000000000000000000| delays: 08+-04
  m1, b04: |00000000000000000111111111000000| delays: 21+-04
  m1, b05: |00000000000000000000000000000011| delays: 00+-02
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b03 delays: 08+-04
  m2, b00: |00000000000000000000000000000000| delays: -
  m2, b01: |00000000000000000000000000000000| delays: -
  m2, b02: |00000000000000000000000000000000| delays: -
  m2, b03: |00111111111000000000000000000000| delays: 06+-04
  m2, b04: |00000000000000111111111000000000| delays: 18+-04
  m2, b05: |00000000000000000000000000011111| delays: 29+-02
  m2, b06: |00000000000000000000000000000000| delays: -
  m2, b07: |00000000000000000000000000000000| delays: -
  best: m2, b03 delays: 06+-04
  m3, b00: |00000000000000000000000000000000| delays: -
  m3, b01: |00000000000000000000000000000000| delays: -
  m3, b02: |10000000000000000000000000000000| delays: 02+-02
  m3, b03: |00001111111111000000000000000000| delays: 09+-05
  m3, b04: |00000000000000001111111111000000| delays: 21+-05
  m3, b05: |00000000000000000000000000000011| delays: 00+-02
  m3, b06: |00000000000000000000000000000000| delays: -
  m3, b07: |00000000000000000000000000000000| delays: -
  best: m3, b04 delays: 21+-05
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
Write: 0x40000000-0x40200000 2.0MiB 
Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 78.8MiB/s
   Read speed: 81.0MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading demo.bin to 0x40000000 (39280 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (9.9KB/s).
[LITEX-TERM] Booting the device.
[LITEX-TERM] Done.
Executing booted program at 0x40000000

--============= Liftoff! ===============--

Nothing after the liftoff.

So can someone tell me how to overcome this problem or how to change in the linker file from SRAM to main_ram ?

enjoy-digital, 2022-06-22:
Hello @FATHY174,

can you share the command you are using for this build? The demo is expecting to have a `main_ram `in the design, if not present in your SoC, you can uses something like `--integrated-main-ram-size=0x10000` to add it to your design.

FATHY174, 2022-06-22:
Hello @enjoy-digital 
I use :
> python3` demo.py --build-path=/home/fathymd/litex2/litex-boards/litex_boards/targets/build/xilinx_vc707/

then access the FPGA with :
>  litex_term /dev/ttyUSB2 --kernel=demo.bin

FATHY174, 2022-06-22:
The main ram is presented in the region from 40000000 to 42000000 and initializing SDRAM but when I change SRAM to the main ram in the linker file, it gets no output after liftoff

enjoy-digital, 2022-06-22:
With the VC707 you shouldn't have to modify the linker script of the demo, it should work unmodified.

enjoy-digital, 2022-06-22:
FATHY174: Maybe you can first modify the SRAM size with `--integrated-sram-size=0x????` and once working, you could see how to move all the code to `main_ram`.

FATHY174, 2022-06-22:
Sorry, I am new to litex, do you have an idea how to store a code in the main_ram(DRAM) instead of SRAM?

FATHY174, 2022-06-27:
@enjoy-digital Is there a way I can store only a header file in the main_ram and use it normally in the litedram cmds?