enjoy-digital/litex #1341: Openocd telnet step debugging stuck on lw instructions

fritzbauer, 2022-06-23:
### Issue
When stepping through the code using the telnet connection to openocd at some point the program counter won't be incremented anymore. I can step over the instruction with "step <next_address>" and manually setting the register according to the lw instruction, however this is tedious.

The issue only occurs when this as cpu0.yaml: https://github.com/enjoy-digital/litex/files/8609336/cpu0.yaml.txt
However, when using this version it is working:
https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/master/pythondata_cpu_vexriscv/verilog/VexRiscv_Debug.yaml

Maybe the wiki needs to be updated accordingly?

### Environment
I am running a plain build of the QMTech 5CEFA2 Board:
`python litex_boards/targets/qmtech_5cefa2.py --cpu-type=vexriscv --cpu-variant=standard+debug --build --load`

Furthermore I followed these instructions for debugging: https://github.com/enjoy-digital/litex/wiki/Use-GDB-with-VexRiscv-CPU

Please find the output of the following command attached:
```
./src/openocd -d 
	-c 'interface dummy' 
	-c 'adapter_khz 1' 
	-c 'jtag newtap lx cpu -irlen 4' 
	-c 'target create lx.cpu0 vexriscv -endian little -chain-position lx.cpu -dbgbase 0xF00F0000'               
	-c 'vexriscv cpuConfigFile cpu0.yaml'               
	-c 'vexriscv networkProtocol etherbone'               
	-c 'init'               
	-c 'reset halt'
``` 
[ocdLog.log](https://github.com/enjoy-digital/litex/files/8968884/ocdLog.log)

Dolu1990, 2022-06-23:
Hi,

> 	Error: 146 97 core.c:1125 jtag_examine_chain_check(): JTAG scan chain interrogation failed: all ones
	Error: 147 97 core.c:1127 jtag_examine_chain_check(): Check JTAG interface, timings, target power, etc.
	Error: 148 97 core.c:1617 jtag_init_inner(): Trying to use configured scan chain anyway...
	Debug: 149 97 core.c:1368 jtag_validate_ircapture(): IR capture validation scan
	Error: 150 97 core.c:1417 jtag_validate_ircapture(): lx.cpu: IR capture error; saw 0x0f not 0x01
	
Seems to indicate an issue at the JTAG level.

But with the working yaml, it doesn't appear and all work well ?

fritzbauer, 2022-06-23:
According to the wiki this is the expected output: https://github.com/enjoy-digital/litex/wiki/Use-GDB-with-VexRiscv-CPU#-example

Just checked, these lines also occur with the working yaml definition.

Also, what I forgot to mention before:
I am using the ocd debugger over the uartbone, but via a dedicated uart, not via crossover as described in the wiki:
`self.add_uartbone(name="serial_debug", baudrate=115200)`

Dolu1990, 2022-06-23:
Ahhh funny XD

> Maybe the wiki needs to be updated accordingly?

I think you are right, the not working yaml is a reaaaaaly old one :)

Dolu1990, 2022-06-27:
Updated, thanks :)

fritzbauer, 2022-06-27:
Nice, thank you!

closed 2022-06-27 by fritzbauer