enjoy-digital/litex #1344: litex_sim generates wrong reset adress when spiflash is present.

Shatur, 2022-06-27:
The following command:

```bash
litex_sim --csr-svd=soc.svd --memory-x=memory.x --with-spi-flash
```

Generates the following `memory.x`:

```
MEMORY {
	rom : ORIGIN = 0x00000000, LENGTH = 0x00020000
	sram : ORIGIN = 0x10000000, LENGTH = 0x00002000
	spiflash : ORIGIN = 0x01000000, LENGTH = 0x01000000
	csr : ORIGIN = 0xf0000000, LENGTH = 0x00010000
}
 
REGION_ALIAS("REGION_TEXT", spiflash);
REGION_ALIAS("REGION_RODATA", spiflash);
REGION_ALIAS("REGION_DATA", sram);
REGION_ALIAS("REGION_BSS", sram);
REGION_ALIAS("REGION_HEAP", sram);
REGION_ALIAS("REGION_STACK", sram);
 
/* CPU reset location. */
_stext = 0x000000;
```

But CPU reset location points 0x00000000, which results in the following linker error:
```
section '.text.dummy' will not fit in region 'spiflash': overflowed by 18446744073675997184 bytes
```

Is this the expected behaviour? Changing it to `0x01000000` compiles. I using the latest litex version at the time of writing.

enjoy-digital, 2022-06-27:
Hi @Shatur,

I just did a test and things are working correctly on my setup. Which version of GCC are you using? (here 10.1.0).

Shatur, 2022-06-27:
> I just did a test and things are working correctly on my setup.

So CPU reset location should point to 0x00000000?

>  Which version of GCC are you using?

I haven't tried it in C. Could you send me your working example?

I tried doing it in Rust, here is the example where I can reproduce the issue:
[hello_world.zip](https://github.com/enjoy-digital/litex/files/8994810/hello_world.zip)

Here is the command I used to generate `memory.x` and `soc.svd` (included in the archive):
```bash
litex_sim --csr-svd=hal/litex_sim_pac/soc.svd --memory-x=hal/litex_sim_pac/memory.x --with-spi-flash --with-sdram --cpu-variant=minimal
```

To build the project simply run `cargo build`. I used this repo as reference: https://github.com/pepijndevos/rust-litex-example. If I put my `memory.x` and `soc.svd` to this repo - I will have the same error. This is why I assumed that there is a bug in litex_sim generation.

enjoy-digital, 2022-06-27:
It seems you are adapting a Rust example to run it in simulation. You'll also have to adapt `litex_sim`: 
- Define the CPU reset address with:` self.cpu.set_reset_address(0xxxxyyyy)`
- Use spi_flash-init to provide SPI Flash contents.

Shatur, 2022-06-27:
> It seems you are adapting a Rust example to run it in simulation.

Yes!

> Define the CPU reset address with: self.cpu.set_reset_address(0xxxxyyyy)

Can I use `--cpu-reset-address` for it? I tried it, it have not effect on generated `memory.x`

> Use spi_flash-init to provide SPI Flash contents.

I need to fill it with my compiled firmware (.bin), right? But before it I need to compile it and I have the mentioned error above :(

enjoy-digital, 2022-06-27:
Sorry I don't have the answer now since the use-case is a combining different contributions to the project that I don't necessarily use (Rust export support, SPI Flash simulation support) and small things will probably have to be adjusted. This would need to be explored butI don't necessarily have the time now. Can you try to explore/adjust thing yourself and report back if you identify ajustements/improvements to make? Thanks.

Shatur, 2022-06-27:
I would like to help, but I need a working starting point. Could you share your working example with GCC (you said previously that you don't have the mentioned linker error)?
My current problem that the generated `memory.x` can't be compiled without modification. So I assume that there is an issue.

enjoy-digital, 2022-06-28:
I would also like to help but would first need to understand what you are trying to do :):
- The command you provided compiles without errors with GCC and the simulation runs correctly.
- litex_sim is intended to be used for simulation.
- The Rust example you are using (and that I'm not familiar with) probably assumes that some specific features are present in the 
SoC.
So a brief description of what you are trying to achieve would help me :)

enjoy-digital, 2022-06-28:
Thanks for the additional info. Do you really need the SPI Flash? Because you can run code from ROM directly and use `--rom-init `to provide a custom binary (eventually compiled with Rust).

The compile issue seems more related to adaptation that would have to be done to the Rust example than an issue with `memory.x` generated by LiteX. I'm unfortunately no familiar with Rust. For the "hello world" in C, as you can see the LiteX BIOS is running correctly, so a hello world would also run correctly.

Shatur, 2022-06-28:
> Thanks for the additional info. Do you really need the SPI Flash? Because you can run code from ROM directly and use --rom-init to provide a custom binary (eventually compiled with Rust).

Yes, this makes sense. The only reason I adding `--with-spi-flash` is because the generated `memory.x` looks in the following way without it:

```
MEMORY {
	rom : ORIGIN = 0x00000000, LENGTH = 0x00020000
	sram : ORIGIN = 0x10000000, LENGTH = 0x00002000
	csr : ORIGIN = 0xf0000000, LENGTH = 0x00010000
}

REGION_ALIAS("REGION_TEXT", spiflash);
REGION_ALIAS("REGION_RODATA", spiflash);
REGION_ALIAS("REGION_DATA", sram);
REGION_ALIAS("REGION_BSS", sram);
REGION_ALIAS("REGION_HEAP", sram);
REGION_ALIAS("REGION_STACK", sram);

/* CPU reset location. */
_stext = 0x000000;
```

An alias for `spiflash` is declared, but the MEMORY section does not contain it. This is why I assumed that BIOS that by default runs from rom doesn't use this file.

Shatur, 2022-06-28:
Discussed with the author of the https://github.com/pepijndevos/rust-litex-example
It's totally fine to modify `memory.x`. So there is no actual issue. Thank you both a lot!

enjoy-digital, 2022-06-28:
Thanks @Shatur, I'll still have a look then since the spiflash should probably not be present in the generated memory.x if the SoC does not have a SPI Flash.

Shatur, 2022-06-28:
Yes, maybe without SPI it's better to generate it like this:
```
MEMORY {
	rom : ORIGIN = 0x00000000, LENGTH = 0x00020000
	sram : ORIGIN = 0x10000000, LENGTH = 0x00002000
	csr : ORIGIN = 0xf0000000, LENGTH = 0x00010000
}

REGION_ALIAS("REGION_TEXT", rom);
REGION_ALIAS("REGION_RODATA", rom);
REGION_ALIAS("REGION_DATA", sram);
REGION_ALIAS("REGION_BSS", sram);
REGION_ALIAS("REGION_HEAP", sram);
REGION_ALIAS("REGION_STACK", sram);

/* CPU reset location. */
_stext = 0x000000;
```

(replace spiflash with rom)