enjoy-digital/litex #1345: Undefined reference to « calloc »

Esteves4, 2022-06-28:
Hi,

I'm using the Litex at commit 1a90549fa345cccd5a792ce09219b142f4d851a1, but it generates an error of undefined reference to « calloc » while building the firmware.

I need to use this commit because of the new tool "litex_soc_gen".

I have tested the program at another target (without using "litex_soc_gen") with a new installation of Litex at the release/tag 2022.04 and everything works smoothly. Then, I changed only the litex repo to the commit 1a90549fa345cccd5a792ce09219b142f4d851a1 and the error shows up.

Esteves4, 2022-06-29:
I saw that in the litex/soc/software/common.mak file, the difference between the two versions of the litex is the -flto flag in the common flags.

I just added the flag again and now I can compile it.

enjoy-digital, 2022-07-05:
Hi @Esteves4,

the issue is probably just that LiteX is not able to detect the `-flto `change and does not trigger the full re-build of the firmware. I'm not sure we this is useful to handle this automatically, so I would just recommend doing a clean build.

closed 2022-07-05 by enjoy-digital