enjoy-digital/litex #1351: Question on SPI Flash Boot

finchlt, 2022-07-01:
LiteX Devs,

We are looking at the linux-on-litex-vexriscv design for a project. So far the litex tools an ecosystem and the overall design has exceeded my expectations on every level. This is so refreshing after building FPGA systems in proprietary tools. 

So far I have been able to build and run custom board designs by extending litex-boards successfully for our custom hardware. 

I have succesfully been able to boot our Linux distro using serialboot and tftp/netboot, however what we really need is SPI Flash boot. On our hardware we have access to SPI Flash memory, and its successful configured and running as part of the soc. 

Additionally looking through the bios codebase i see that there is a flashboot configuration. Also I see on the Wiki at the followomg link [https://github.com/enjoy-digital/litex/wiki/Load-Application-Code-To-CPU](https://github.com/enjoy-digital/litex/wiki/Load-Application-Code-To-CPU) that there is plans to support and/or document spiflash boot. 

My plan is to evaluate the following:
1. Load a opensbi/u-boot binary from spiflash to ram, run the boot sequence through bios -> opensbi -> u-boot payload -> load linux initrd from spiflash with u-boot
2.  Load entire image from spifash to ram including opensbi and the linux initrd, then boot through bios -> opensbi -> kernel directly

I had the following questions:
1. Am I on the right track with bios flashboot bios code or is there something else in mind with the future SPI flash boot process?
2. It looks like the flashboot code in the bios calls check_image_in_flash which looks for a header on the flash image containing the image size at the baseaddress (FLASH_BOOT_ADDRESS) and a crc of the image at base address + 4. Does this sound right? Is so I can add this information into the data written to flash easily. 
3. Anything else I might be missing here?

Additionally were happy to implement new features if needed as well.

Thanks,

Logan

enjoy-digital, 2022-07-04:
Hi @finchlt,

thanks for the feedback. You are on the right track, but we should also document and add an example for boot from SPI Flash. I'll first do this and will then give you more info if needed.

finchlt, 2022-07-05:
@enjoy-digital Thanks so much for the additional update.  This project is really great.