enjoy-digital/litex #1377: Where is the CVA6 wrapper?

sergachev, 2022-07-19:
CVA6 support requires a couple of files:
https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/cva6/core.py#L156

An attempt to run `litex_sim --cpu-type=cva6` reports Flist.cv64a6_imafdc_sv39 as missing - this is easy to fix by prefixing its path with `core/` - but Flist.cva6_wrapper is missing completely and I can not find any trace of it anywhere. It is not in https://github.com/litex-hub/pythondata-cpu-cva6; not in https://github.com/suppamax/pythondata-cpu-cva6 either; also not in PyPi built packages; links to it in https://github.com/enjoy-digital/litex/pull/1294 are broken. I can not find it even in the history of the repos.
What is the story with it?

@suppamax @mithro

enjoy-digital, 2022-07-20:
Hi @sergachev,

the wrapper seems to be there now: https://github.com/suppamax/pythondata-cpu-cva6/commits/cva6_wrapper
As I suggested to @suppamax, it would probably be better to have the wrapper directly integrated in LiteX as we are doing for other cores that require it. @sergachev: Feel free to go further in this direction if you are able to get things working with this branch.

suppamax, 2022-07-20:
Hi @sergachev 

you can check [this PR](https://github.com/litex-hub/pythondata-cpu-cva6/pull/1)

Anyway, as @enjoy-digital already mentioned, the plan is to stop using the flist and integrate it in LiteX

enjoy-digital, 2022-07-20:
Using the flist would still be fine (we are doing it for some other CPUs) but having the specific flist + wrapper directly in LiteX would probably simplify things.

mithro, 2022-07-20:
As the `pythondata-cpu-cva6` is auto-generated, I recommend we include it directly in LiteX repository like @enjoy-digital suggests.