enjoy-digital/litex #1382: Printing issue

FATHY174, 2022-07-21:
Hello @enjoy-digital 

I just want to know why printing any float number in the C codes gives the output `*float*` or even gives `undefined reference to `__extendsfdf2'` is there any library I should include or there is something I don't get?

FATHY174, 2022-07-26:
@enjoy-digital 
adding something like this to any C file:
`double uu = 5.9;

	int kkk =3;
        printf("\nuu = %f\n",uu);
        double ll = 2.9;
        printf("\nll = %f\n",ll);
        double tt = (uu*(double)kkk)/ll;
        printf("\ntt = %f\n",(uu*(double)kkk)/ll);
        printf("\ntt dec = %d\n",(int)((uu*(double)kkk)/ll));`

gives :
`uu = *float*'
ll = *float*'
rtt = *float*`

what's the reason??

FATHY174, 2022-08-02:
I found the problem is related to the PICOLIBC :
flags for floating should be enabled

closed 2022-08-02 by FATHY174

enjoy-digital, 2022-08-02:
Good @FATHY174, would you mind sharing the solution? We'll try to integrate it.

tcal-x, 2022-08-02:
I had always assumed that `printf` `%f`/`%d` support was left out for reasons of code size, and that's why you'd do something simple if you needed one or two decimals of accuracy, as here: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/memtest.c#L168-L177 .

But yes it would be nice to have an option for having floats directly supported by printf!

befedo, 2023-03-27:
Hi there,

I'm also looking for a way to `printf` floats (using the XADC module) and reading through the [picolibc doc](https://github.com/picolibc/picolibc/blob/main/doc/printf.md) doesn't seem to work. Setting `PICOLIBC_FLOAT_PRINTF_SCANF` also yields `*float*`. Any hints on how to solve this?

Thanks for your time and best regards
befedo

befedo, 2023-03-27:
> Hi there,
>
> I'm also looking for a way to `printf` floats (using the XADC module) and reading through the [picolibc doc](https://github.com/picolibc/picolibc/blob/main/doc/printf.md) doesn't seem to [...]

Found it ... one could use
```
CFLAGS += -Wl,--defsym=vfprintf=__d_vfprintf -Wl,--defsym=vfscanf=__d_vfscanf
```
don't know why `PICOLIBC_FLOAT_PRINTF_SCANF` isn't sufficient.