enjoy-digital/litex #1396: No such command: read_vhdl

nyaasen, 2022-08-11:
Hi Everyone,

I am trying to build microwatt for the Radiona ULX3S. I run into an error during Yosys execution. Mind you, it works fine on my Basys3 board with Vivado.

Here is what I have tried to address the issue:
Install yosys-ghdl-plugin
- It did not show the required read_vhdl command

Compile yosys vhdl plugin
- https://github.com/YosysHQ/yosys-plugins/issues/7 is the same issue I had gotten, so I tried the following, since it was the same issue I am having on Fedora: https://github.com/ldoolitt/vhd2vl/commit/cfc3304963dcc3fe9284d1122dc5e604a92197e1

Now the error I am getting is `error: no member named 'linenum' in 'Yosys::AST::AstNode'`
This is when I figured I am probably going down the wrong path. How are you guys getting "vhdl_read" to show up during microwatt generation?

Thanks

enjoy-digital, 2022-08-23:
Hi @nyaasen,

to build/use Microwatt with Yosys, we are first converting it to VHDL. This is done here:
https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/microwatt/core.py#L316-L347
and you have to add `+ghdl` to the CPU variant to enable it (ex `--cpu-type=standard+ghdl`)

We are planning to simplify this and automatically detect when VHDL -> Verilog conversion has to be done, but this is not yet implemented.

Vivado accepts VHDL files, that's why it is working correctly on your Basys3 build.

Can you try with this?

Thanks.

nyaasen, 2022-08-25:
I went ahead and did it. It worked for --cpu-variant=standard+ghdl, rather than --cpu-type.

Thanks for the help!

closed 2022-08-25 by nyaasen

Haron123, 2024-08-16:
The named cpu-variant doesnt seem to exist

Executing ``./colognechip_gatemate_evb.py --build --cpu-variant=standard+ghdl`` gives me

```
ERROR:SoC:standard+ghdl CPU variant not supported, supported are: 
 - full
 - full+cfu
 - full+cfu+debug
 - full+debug
 - imac
 - imac+debug
 - linux
 - linux+debug
 - linux+no-dsp
 - lite
 - lite+debug
 - lite+debug+hwbp
 - minimal
 - minimal+debug
 - minimal+debug+hwbp
 - secure
 - secure+debug
 - standard
 - standard+debug
```

and if i dont use it i get the yosys msg

```
 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.42+15 (git sha1 74a1dd99a, g++ 14.1.1 -Os)

-- Executing script file `colognechip_gatemate_evb.ys' --
ERROR: No such command: read_vhdl (type 'help' for a command overview)
```

Can anyone help with a solution to this? I really wanna use my VHDL Cores in Litex.

trabucayre, 2024-08-17:
`--cpu-variant=standard+ghdl` is only limited to microwatt CPU (to force the vhdl to verilog conversion): by default LiteX build a vexriscv softcore: this why this value isn't recognize/supported.

With *yosys* you cannot directly mix vhdl and verilog files: is your core available somewhere?

The [can core](https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/can/ctu_can_fd.py) is an example on how to integrate vhdl code in a verilog project using ghdl: by using `VHD2VConverter` instead of `Instance` your vhdl is automatically converted to a verilog code at build time and the resulting file is integrated to the project before synthesis.

Haron123, 2024-08-17:
Thank you, i managed to get it to work!