enjoy-digital/litex #1404: Udp stream problem

ramalmar, 2022-08-22:
Hello,

I am trying to continuously stream various signals from acquisition chips through the SPI port. The data consists of 1536 bytes with a sample rate that can be changed from 1 to 32 kHz. The information is captured in each spi clock cycle and sent through the LiteEthStream2UDPTX module when 1536 bytes are reached.

        udp_streamer = LiteEthStream2UDPTX(
            ip_address=convert_ip(host_ip),
            udp_port=host_udp_port,
            fifo_depth=1536,  
            send_level=1536  
        )

The system acquires perfectly at 8 kHz, however when I increase the frequency to 16 kHz something strange happens with the FIFO memory. Which at a random moment of the acquisition is not emptied and corrupts the rest of the transmission.

**8 khz**
[image]

**16Khz**

[image]
Time when the error occurs

[16khz_litescope.zip](https://github.com/enjoy-digital/litex/files/9392700/16khz_litescope.zip)

After the error, the fifo.level variable does not return to 0 but remains at 5. This means that the first 5 bytes of a sample transmitted are actually the last 5 bytes of the previous sample. Corrupting the rest of the acquisition
[image]

Observing the acquisition with litexCope I have assumed that the system has enough time to carry out the process at 16khz, however I think that by increasing the frequency I have exceeded some unknown limit. Are there any other variables that I should take into consideration?

Thank you very much in advance

enjoy-digital, 2022-08-23:
Hi @ramalmar,

the code has been reworked a bit last month and `send_level` parameter has been removed. You can now just set `last` signal at the end of the packet and the packet will be transmitted. Can you update LiteX/LiteEth and test with this?

If still not working, could you redo some captures?

Thanks

ramalmar, 2022-08-24:
Hi @enjoy-digital,
I updated the LiteEthe and removed the send level parameter and now it seems to work well even at 32 khz!. 
However, I do not send the data with the` last` signal but when the FIFO is full. This way works but would it be better to use the `last` signal?

Thanks!

enjoy-digital, 2022-08-24:
Hi @ramalmar,

this will also be fine like this if you are streaming continuous data.

ramalmar, 2022-08-29:
Hi @enjoy-digital,

It thought it was working fine but it seems I made a mistake checking the signal and the error in the FIFO is still there, but now the fifo.level stays at the end at 4 instead of 5 :-(

```
        udp_streamer = LiteEthStream2UDPTX(
            ip_address=convert_ip(host_ip),
            udp_port=host_udp_port,
            fifo_depth=1727, 
        )
```
[image]
[capture_subsam_4.zip](https://github.com/enjoy-digital/litex/files/9445104/capture_subsam_4.zip)

In a random sample, ` source.ready` goes low and the FIFO content is not sent until the next sample..it usually happens when I do readings of more than 5 secons at 16 khz. It looks like it crashed.

The capture is subsampled by 4 to see multiple data samples. If necessary I can try to capture a specific moment or show more signs:)

Thanks