enjoy-digital/litex #1407: NaxRiscv simulator hangs at memtest

boondockenergy, 2022-08-28:
Trying NaxRiscv using `litex_sim --cpu-type naxriscv --integrated-main-ram-size 1024`

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Aug 28 2022 13:38:09
 BIOS CRC passed (b29f8421)

 LiteX git sha1: 50a5e137

--=============== SoC ==================--
CPU:		NaxRiscv @ 1MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
MAIN-RAM:	1KiB

--========== Initialization ============--
Memtest at 0x40000000 (1.0KiB)...
```

cklarhorst, 2022-08-29:
Hi @boondockenergy,
there is already a PR that should fix this (#1355).
Maybe @enjoy-digital or @Dolu1990 could review it?

Dolu1990, 2022-08-29:
Yes right, that is the issue, continueing on #1355

Dolu1990, 2022-09-12:
Merged

closed 2022-09-12 by Dolu1990